convertersystems/opc-ua-client #97: cannot writeAsync

gvfbb, 2019-10-07:
hello

using Workstation.UAClient I have no problems in reading values, using ReadAsync method.
 I am also  trying to use WriteAsync method in order to edit a server value, but I do not succeed. The tag I wish to change is Int16 type.

This is the code:

        static async Task Main(string[] args)
        {
            var clientDescription = new ApplicationDescription
            {
                ApplicationName = "Workstation.UaClient.FeatureTests",
                ApplicationUri = $"urn:{System.Net.Dns.GetHostName()}:Workstation.UaClient.FeatureTests",
                ApplicationType = ApplicationType.Client
            };

            var channel = new UaTcpSessionChannel(
            clientDescription,
            null, // no x509 certificates
            new AnonymousIdentity(), // no user identity
            "opc.tcp://localhost:4841", 
            SecurityPolicyUris.None); // no encryption

            try
            {
                // try opening a session and reading a few nodes.
                await channel.OpenAsync();

                var writeRequest = new WriteRequest
                {
                    NodesToWrite = new[]
                        {
                            new WriteValue
                            {
                                NodeId = NodeId.Parse("ns=4;s=Address Space.Dev00.Tag001"),
                                AttributeId = AttributeIds.Value,
                                Value = new DataValue(100)                                
                            }
                        }
                };

                var writeResult = await channel.WriteAsync(writeRequest);
                string sRes = writeResult.Results[0].ToString();

                await channel.CloseAsync();
            }
            catch (Exception ex)
            {
                await channel.AbortAsync();
                Console.WriteLine(ex.Message);
            }
        }

Can you help me please?

Thanks in advance 

Giovanni

gvfbb, 2019-10-07:
hello

I succeeded with the following conversion:

            var writeRequest = new WriteRequest
            {
                NodesToWrite = new[]
                    {
                        new WriteValue
                        {
                            NodeId = NodeId.Parse("ns=4;s=Address Space.Dev00.Tag001"),
                            AttributeId = AttributeIds.Value,
                            Value = new DataValue(Convert.ToUInt16(100) )                               
                        }
                    }
            };

is it the correct way?

awcullen, 2019-10-07:
You are exactly correct.  Most servers require that the WriteValue be an instance of the exact type expected.  So, int and UInt16 are different and needs to be converted beforehand.
Well done.

gvfbb, 2019-10-07:
thanks a lot for making available your code and dlls, and for your quick answers on this forum

closed 2019-10-11 by awcullen