convertersystems/opc-ua-client #100: Allow to dispose open TCP Sessions

mallibone, 2019-10-08:
I have noticed that when using the `UaTcpSessionChannel` one can't dispose of the object. Instead (if I read the code correctly) one should always invoke `CloseAsync` to ensure the TCP socket is freed up. Would it be possible to inherit from IDisposable in the `ICommunicationObject` to allow users of the library to also dispose of the no longer required object? Or would this bring any inconvenience at other points?

By the way - nice library 👍

quinmars, 2019-10-08:
That's not possible because `IDisposable` is purely synchronous, whereas `CloseAsync` is an async method, i.e., the result needs to be awaited. With C#8 and netstandard 2.1 there is the asynchronous interface [`IAsyncDisposable`](https://docs.microsoft.com/en-us/dotnet/api/system.iasyncdisposable?view=dotnet-plat-ext-3.0). But netstandard 2.1 will only be supported by dotnetcore 3 and not by the .Net-Framework. Maybe there will be a nuget package providing that interface. I hope that dropping support for the .Net-Framework is not an option at the moment.

mallibone, 2019-10-08:
Yes it is true `IDisposable` is synchronous - but then again invoking [Dispose/Close](https://github.com/convertersystems/opc-ua-client/blob/adcfa20f31280c8bf03646ef4f53b01978ae01c6/UaClient/ServiceModel/Ua/Channels/UaTcpTransportChannel.cs#L266) are also synchronous operations, so there is no need to close the socket in an async fashion as far as I understand?

Or perhaps my question should be: if the channel faults and while being registered to the event handler for closed/faulted - how should I close/dispose of the open socket? Do I have to abort the connection in case of a fault?

quinmars, 2019-10-09:
> Or perhaps my question should be: if the channel faults and while being registered to the event handler for closed/faulted - how should I close/dispose of the open socket? Do I have to abort the connection in case of a fault?

`IDisposable.Dispose()` should usally be called after you finished using the object. But inside of an event handler the object is still executing code, else it couldn't call the delegate. Hence, I don't think this is the right place to dispose the object.

Your question, however, is valid, in particular for subscriptions, where you do not have the classic open-call-dispose pattern. Unfortunately I don't have an answer for that. Maybe @awcullen can help.

mallibone, 2019-10-11:
@quinmars yes you are right that we are in an event handler. Nonetheless, if I receive the channel is faulted and want to start a new connection. I will have to call `Abort` which will then dispose of the underlying TcpSocket. If I understand #44, correctly one has to recreate the `UaTcpTransportChannel` anyway?

@dojo90 will have to check out that PR 🙂

awcullen, 2019-10-11:
Good discussion. When building this library I tried to stay as close to the design of the WCF channels as I could.  You will notice that I borrowed the base CommunicationObject from https://docs.microsoft.com/en-us/dotnet/api/system.servicemodel.channels.communicationobject?view=netframework-4.8  And so what works well for WCF should also apply to UaClient.  For instance, there is advice to avoid 'using' a channel, because exceptions can happen at the '}' when Dispose is called (which forwards to Close in the MSFT version). Please see the discussion and examples https://docs.microsoft.com/en-us/dotnet/framework/wcf/samples/use-close-abort-release-wcf-client-resources 
You may wonder why I did not implement Dispose forwarding to Close. As @quinmars points out, I wanted Close to be async (because it sends two rounds of network msgs) and AsyncDisposable hadn't been invented yet!

awcullen, 2019-10-14:
To summarize so far:
1) if you get an exception during OpenAsync, any service call, or CloseAsync, you should try catch it and call AbortAsync, which closes the socket.
2) Another way to go is to add a handler to channel.Faulted that calls AbortAsync.
3) Channels should not be reused or reopened, if it has faulted or even if it closed successfully.

closed 2019-10-19 by awcullen