convertersystems/opc-ua-client #103: HowTo handle "faulted" connection attempt

dojo90, 2019-10-10:
Actually I want to implement an auto reconnect on `faulted` connection attempts.

After the connection failed, the actual `State = CommunicationState.Faulted`. 
In `OpenAsync()` (https://github.com/convertersystems/opc-ua-client/blob/master/UaClient/ServiceModel/Ua/Channels/CommunicationObject.cs#L187) there is a check if the `State == CommunicationState.Faulted` (https://github.com/convertersystems/opc-ua-client/blob/master/UaClient/ServiceModel/Ua/Channels/CommunicationObject.cs#L473). This results into an `InvalidOperationException("Channel not modifiable.")` .

How should I resolve the problem?

`Close()` is not possible to be called -> `Exception(Channel faulted)`

dojo90, 2019-10-10:
@quinmars thanks for your reply. I also saw this issue already and it's solution, but I don't think that this is a nice way how to handle lost or interrupted connections. I will try to recreate the connection logic in the library and make a PR. I my opinion it's much nicer, if the subscriptions can last as they are and only the session layer will be replaced with a new and valid one.

mallibone, 2019-10-11:
@dojo90 I am trying to do the same thing 😆 hence my question in the issue #100 - my solution is to create a new UaTcpTransportChannel on every connect/reconnect.
But you will want to dispose of the faulted TcpSocket first. For this call `Abort` which disposes of the currently faulted [channel](https://github.com/convertersystems/opc-ua-client/blob/1e7d60da59b256fdd2f9c604fb8bd69d049a5e34/UaClient/ServiceModel/Ua/Channels/UaTcpTransportChannel.cs#L277). It does not check for the connection state and but disposes of the TcpSocket beneath.
I don't know if it is always the desired effect to stay connected even if a connection interrupts. But would, of course, be nice if one could configure the connection to reconnect automatically should an error occur.

awcullen, 2019-10-14:
I like what you have done.
You may wish to add to channel.Faulted handler to Abort the channel (which closes the socket without further network calls)
```
_Channels.Remove(channel);
try
{
     channel.AbortAsync().Wait();
}
catch
{
}
```

closed 2019-10-19 by awcullen

dubeg, 2021-02-19:
@awcullen Hi Andrew. Why would AbortAsync be needed if the channel faulted? Should it already be closed?

awcullen, 2021-02-20:
Back when I started this project, I tried to follow Microsoft's  WCF guidelines.  When you used a WCF client, and there was an exception, you would catch the exception and Abort (vs Close) the Client.
Abort disposes the channel without communicating to the server ( which probably has closed it's session anyway)