convertersystems/opc-ua-client #111: Discussion: Increasing test coverage

quinmars, 2019-10-21:
With the current unit tests we cover around 37% of the code. I estimate that we can write additional tests to reach around 50%. A large and very important part remains uncovered, namely everything that is related to the UA TCP stack. One would need to replace the `TcpClient` by something else. That is a very similar intention as to support different OPC UA transport profiles, like for example Https. So let me first talk about the later, I will come back to the unit tests topic.

I can think of two ways for implementing different OPC UA profiles without writing everything from scratch. I'll present you both of them, but without revealing too much, I strongly prefer the second one.

## A stack of layers

When I started to read about OPC UA and its layers for transport, security and encoding, I thought supporting several profiles means putting the different layers on each other like you would do to make your favorite sandwich. When I than saw your tcp profile stack it totally made sense, because (almost) every layer is a level of inheritance. There is just one major problem with that. You cannot exchange one level with a different level in C#. If inheritance is not flexible enough, the text book answer is to use composition. I thought of it like building a pipeline like in LINQ or Rx. I already started to play around with that idea. But it turns out that you need for every layer an own interface probably two because the information flow does in both directions: up and down. Even worse all the protected methods becoming public because C# interfaces do not support protected members. So the whole stack needs to be hidden in a containing class (let's say `SessionChannel`) so that the consumer does not accidently calls one of those implementation serving methods. As much as a like the sandwich idea, I think it'd become very complicated. On the other hand it keeps you all opportunities.

## Service provider

The second possiblity is probably the more obvious one: injecting the wanted functionality via service provider, for example `ITransport` and `IEncryption`. For the encoding we already have two interfaces. The art will be to make those interfaces general enough, so that they will work for a variety of different implementations. To be honest, they are countable, for the transport it is tcp, https and websockets and for transport security there is only UASC as far as I can see. Currently the
configuration of the transport and security is done directly on the channel. If those are implementation specific they could go with the service provider, e.g.,

```csharp
interface ITransportProvider
{
    ITransport CreateTransport();
}
```
and

```csharp
class TcpTransportProvider
{
    ITransport CreateTransport();

    public uint LocalReceiveBufferSize { get; set; }
    public uint LocalSendBufferSize { get; set; }
    // etc.
}
```

Although some of them are also used by the secure channel layer. At the end I do not have enough insight into the whole story: if the layers are really so loosely connected as presented in the front matter and I do not know if you are interested in supporting more than tcp.

Coming back to unit tests. Sorry for the lengthy interlude. Even if we would go one of those two ways, it be difficult to write unit tests for the whole code. We would need much of the server code to mock a plausible server. And if we really go that way, we should test the already good working code before we change and break it. Hence, no matter if we take one of the two options or 
if we stay with tcp only we need automated integration tests. Unfortunately we cannot use Unified Automations test server in a public CI pipeline. I already look for open source test servers and some projects have them. For example, [FreeOpcUa Python](https://github.com/FreeOpcUa/python-opcua), [Opc Ua Rust](https://github.com/locka99/opcua), [open62541](https://open62541.org/) and presumably also [eclipse milo](https://github.com/eclipse/milo). Some of them have even ubuntu packages! Maybe the tests can be orientated to the certification tests. Unfortunately I do not have access to them, but: http://opcfoundation-onlineapplications.org/ProfileReporting/index.htm lists some of them.

There is one more thing we can do to increase the testable code portion. Some of the main classes like `UaApplication`, `BinaryEncoder`, `BinaryDecoder` or `BaseSubscription` hold direct or indirect references to `UaTcpSecureChannel` or `UaTcpSessionChannel`. Instead of those concrete references, interface could be used. For example:

```csharp
interface ISessionChannel : ICommunicationObject, IRequestChannel, ISourceBlock<PublishResponse>, IObservable<PublishResponse>
{

}
```
could be used instead of `UaTcpSessionChannel` in `UaApplication` and `SubscriptionBase`. Ideally all occurrences of `UaTcpXxx` in the base folder would be replaced by interfaces. Except `UaApplicationBuilder` at some point you have to get concrete. Those would be breaking changes and should be delayed for a 3.0 version.

My plans for the near future are to increase the unit tests coverage and than start to implement automated integration tests. If you have any experience with one of the test servers I'd be highly interrested to hear from your impressions.

awcullen, 2019-10-23:
I would start with open62541. It is the only one that has passed certification. 
ISessionChannel looks like a good idea. 
In order to use composition at all levels, then SessionChannel would hold a SecureChannel and SecureChannel would hold a TransportChannel.  I would code SessionChannel.Open() to call SecureChannel.OnOpening() (which calls TransportChannel.OnOpening) then SecureChannel.OnOpen() and SecureChannel.OnOpened().
Is that the idea, to avoid inheritance?

quinmars, 2019-10-23:
> Is that the idea, to avoid inheritance?

Almost. The facade class, which would have a similar public API like `UaTcpSessionChannel` would still inherit from `CommunicationObject`. If `OnOpen` is called it would call `ISessionChannel.OnOpen`, which would call `ISecureChannel.OnOpen`, which would call `ITransportChannel.OnOpen`. There would be a similar chain for `OnOpening`, `OnClose`, etc.

quinmars, 2020-01-06:
A short update: With PR #126 we reach the 50% milestone. There is still place to increase coverage with plain unit tests. Namely for the monitored items, `EventHelper`, `ArraySegmentExtensions`, `DataValueExtensions`, `VariantExtensions`, `ServiceOperation`. 

For the integration tests test, I still waiting on the open62541 team to ship a runable test server.

dojo90, 2020-02-13:
Can someone explain to me what the advantage of many layers is? To be honest, I find it very difficult to keep track (within the source code).

For my personal needs i have wrote a wrapper that contains a `UaTcpSessionChannel`. Is there a use case for which I also need the other channels (`UaTcpSecureChannel`, `UaTcpTransportChannel`)?

I am also very positive about refactoring, as there are some differences in the structure of the `StatusCode` of the `ResponseClasses`.

- `WriteResponse.Results[]`
  - Results[] is of type `StatusCode`
- `BrowseNextResponse.Results[].StatusCode`
  - Results[] is of type `BrowseResult` where a `StatusCode` property is defined
- `BrowseResponse.Results[].StatusCode`
  - Results[] is of type `BrowseResult` where a `StatusCode` property is defined
- `AddNodesResponse.Results[].StatusCode` 
  - Results[] is of type `AddNodesResult` where a `StatusCode` property is defined
..

I think it would be much better if an `interface` would be defined for it, so that you only have to write one `extensionMethod` to process a `XYZResponse`.

Even if this misses the actual topic of the issue, I wanted to mention it, since the implementation of `dependency injection` and `interfaces` was discussed before.

quinmars, 2020-02-14:
> Can someone explain to me what the advantage of many layers is? To be honest, I find it very difficult to keep track (within the source code).

Seperation of concerns. The 3 classes are already rather large. In particular the `UaTcpSecureChannel` with its nearly 2000 lines is hard to grasp. Merging the 3 classes to one class would end in a large mess. I do not see how that would make it easier to understand the code.

The OPC UA specification also describes those three layers separately, so it totatly makes sense to handle them in code separately. As discussed above, it is another topic if inheritance is the right tool for the separation. At the moment I think it is the best choice.

> Is there a use case for which I also need the other channels (UaTcpSecureChannel, UaTcpTransportChannel)?

Probably not directly, but `UaTcpDiscoveryService` is using an `UaTcpSecureChannel` internally.

> I am also very positive about refactoring, as there are some differences in the structure of the StatusCode of the ResponseClasses.

The response classes are defined by the OPC UA specification. Andrew cannot change them.

> I think it would be much better if an interface would be defined for it, so that you only have to write one extensionMethod to process a XYZResponse.

I wonder what the purpose of such an extension method would be. But you could easily write an extension method to get the status codes of a service request:

```csharp
public static StatusCode[] GetStatusCodes(this IServiceResponse  response) => response switch
{
    WriteResponse wr => wr.Results,
    BrowseResponse br => br.Results.Select(r => r.StatusCode).ToArray(),
    ...
}
```

quinmars, 2020-06-29:
I started to work on an [OPC UA server](https://github.com/quinmars/UaTestServer), which is a fork of the Quickstarter example of the OPC foundation. The original one is actually a WinForms application, but I changed it to a dotnet core console application, thus it will be possible to integrate it in the CI. Since I've got the tests now passing with the UaCPPServer of unified automation, I'll try to change my UaTestServer so that the tests pass with it, too. It should work as an drop-in replacement. At least that's the plan.

quinmars, 2021-12-23:
The discussion will continue in #218.

closed 2021-12-23 by quinmars