convertersystems/opc-ua-client #112: Can't OpenAsync channel in Xamarin App

ForgotUsername1, 2019-10-22:
I used examle code located in ReadMe. In Console App it works correctly, I can read status from My B&R server- everything works perfect.
I wanted to do the same in Xamarin.Android, but I can't open channel. Every time when I call OpenAsync(), go inside and try catch the exception in  "async Task<T> WithTimeoutAfter<T>".

Bellow I attached gif presenting issue. I will be grateful if you help me.
https://i.imgur.com/gpte6Ll.gifv

awcullen, 2019-10-22:
You will want to check the endpointUrl for the server.
The gif seems to show opc.tcp://localhost:4840
You will have better luck with server ip address, rather than localhost.

ForgotUsername1, 2019-10-23:
Doesn't matter local host or 127.0.0.1
In console app I have also local host in uri and it works fine. 

BTW Gif made by sharex

awcullen, 2019-10-23:
The android simulator is running in a vm and the vm has a separate ip address from the host.
You need to find the ip address of your host and use that.

closed 2019-10-26 by ForgotUsername1