convertersystems/opc-ua-client #119: Nullable rollout phase roadmap

quinmars, 2019-11-28:
In a year .NET 5 will come and [Mads Torgersen advices](https://devblogs.microsoft.com/dotnet/embracing-nullable-reference-types/) to library authors to adopt nullable reference types (NRT) in the next year so that the consumers can benefit from a fully annotated library once they switch to .NET 5.

The [recommended way](https://docs.microsoft.com/en-us/dotnet/csharp/tutorials/upgrade-to-nullable-references) to migrate to NRTs is to switch the language version to C#8, add netcore3.1 as a second target framework beyond .netstandard2.0 and than activate NRTs on a per file basis via the `#nullable enable` switch. I actually don't expect the process to be complicated, but it will probably take around a month to fully migrate opc-ua-client. So the main questions is if you have any plans to make a release in the next time? Because I think the transition should be done between two releases, so you do not ship out a partial annotated library. Alternatively the work can be done in a separate branch.

What do you think?

awcullen, 2019-12-01:
I'll make a branch for this work. Let's plan for a  v3.0 release in the new year.

quinmars, 2019-12-02:
Great!

closed 2019-12-02 by quinmars