convertersystems/opc-ua-client #122: using multiple UaTcpSessionChannels to one server

dojo90, 2019-12-12:
I have a `SIMATIC S7-1500` with an OPC-UA Server running.

I want to discover a specific part of the PLC (`DataBlocksInstance`). To get more speed on discovering, I create a new `UaTcpSessionChannel` for each `ReferenceDescription` found. So (for example) I have `1 main connection` and `5 new child connections` which discover parallel the server.

But here are some problems:
- The events `Opening`, `Opened`, `Faulted`, `Closing`, `Closed` get not fired on the new `child connections`
- I get the following error on the child connections `The session cannot be used because ActivateSession has not been called`

Is there something special about using parallel connections?

awcullen, 2019-12-15:
I was successful in using parallel channels to browse a test server.  Please remember to call:
```
await channel.OpenAsync();
```
Please also see these suggestions to increase performance of Simatic S7 opc ua servers:
https://support.industry.siemens.com/cs/document/109766727/how-can-you-optimize-the-performance-of-the-opc-ua-server-of-a-simatic-s7-1500-and-reduce-the-load-on-the-server-?dti=0&lc=en-WW

Also see this link for advice for opc ua clients:
https://support.industry.siemens.com/cs/document/109765509/how-do-you-best-configure-an-opc-ua-client-to-communicate-efficiently-with-the-opc-ua-server-of-a-simatic-s7-1500-cpu-and-with-the-best-performance-possible-?dti=0&lc=en-WW

dojo90, 2019-12-16:
Thanks for your reply! I have already reduced the amount of tags to a minimum on my OPC Server.

The main problem is the connecting time of the `child connections`. While the first connection was established in 3s, the following connections need 13s. Can you understand that?

My goal is to discover the complete OPC Server as fast as possible. But actually it is faster to use only the main (first) connection for discovering, instead of using parallel connections. I don't saw any other tweaks to speed up the OPC Server in the `TIA Portal`. The maximum sessions is set to `10`, and actually I'm using `6`.

[image]

The following method is called as soon as the `main connection` is `opened`.

```csharp
private List<UaTcpSessionChannel> _GetChannels(UaTcpSessionChannel channel, int count)
{
	List<UaTcpSessionChannel> channels = new List<UaTcpSessionChannel>();
	for (int i = 0; i < count; i++)
	{
		var newChannel = new UaTcpSessionChannel(channel.LocalDescription,  channel.CertificateStore, _UserIdentityProvider, channel.RemoteEndpoint.EndpointUrl);
		channels.Add(newChannel);

		int nr = i + 1;
		Stopwatch stopwatch = Stopwatch.StartNew();

		newChannel.Opened += (sender, args) =>
		{
			stopwatch.Stop();
			Logger.Info($"discovering channel {nr} connected in {stopwatch.Elapsed.Seconds}s");    
		};
		newChannel.Opening += (sender, args) => { };
		newChannel.Faulted += async (sender, args) => { await newChannel.AbortAsync(default); };
		newChannel.Closing += (sender, args) => { };
		newChannel.Closed += (sender, args) => { };
		newChannel.OpenAsync();
	}

	return channels;
}
```

dojo90, 2019-12-16:
I think it's an internal locking problem. I called the `_GetChannels` method with different values:
- `count = 1` - connected in `2s`
- `count = 2` - connected in `5s`
- `count = 3` - connected in `7s`

quinmars, 2019-12-16:
My experience is that the call itself to a remote service is the most expensive part, may it be ADS, the old S7 protocol or OPC UA. I think the most performance improvement you can get is to reduce those calls.

```csharp
await _BrowseDataBlocks(newChannel, _DataBlocksInstanceQueue.Dequeue(), "-");
```

looks like you have a queue where you browse each single node. You could instead add the whole queue/list to `NodesToBrowse`.

dojo90, 2019-12-18:
Maybe I'm using the library in a wrong way, but this is what I'm doing (as soon I'm connected):

1. Browse all `DataBlocksInstance` of my `SIMATIC S7-1500` OPC Server. To get all `instance data blocks` of my plc.
2. Every `ReferenceDescription` of my `data block` is now insert into the `_BrowseDataBlocks` method to scan the `static` part and look for matching UDTs
3. If an UDT is found (`IsOpcStructOrDatablock`) some internal magic happens to automatically create subscriptions and so on..

What I want in general is to improove the speed of the complete process. @quinmars can you give me a hint what I should edit to get better performance?

```csharp
var nodeId = new ExpandedNodeId(new NodeId("DataBlocksInstance", 3), null, 0);
var browseResponse = await channel.BrowseAsync(nodeId);
foreach (ReferenceDescription reference in browseResponse)
{
	if (reference.IsDatablock(out string name))
	{
		await _BrowseDataBlocks(channel, reference);
	}
}
```
```csharp
public static Task<List<ReferenceDescription>> BrowseStaticAsync(this UaTcpSecureChannel channel, ReferenceDescription referenceDescription)
{
	return BrowseAsync(channel, ExpandedNodeId.Parse($"{referenceDescription.NodeId}.Static"));
}

public static async Task<List<ReferenceDescription>> BrowseAsync(this UaTcpSecureChannel channel, ExpandedNodeId nodeId)
{
	var browseRequest = new BrowseRequest
	{
		NodesToBrowse = new BrowseDescription[]
		{
			new BrowseDescription
			{
				NodeId = ExpandedNodeId.ToNodeId(nodeId, channel.NamespaceUris),
				BrowseDirection = BrowseDirection.Forward,
				ReferenceTypeId = NodeId.Parse(ReferenceTypeIds.HierarchicalReferences),
				NodeClassMask = 0u,
				IncludeSubtypes = true, ResultMask = (uint) BrowseResultMask.All
			}
		},
	};
	return await channel._BrowseAsync(browseRequest);
}

private static async Task<List<ReferenceDescription>> _BrowseAsync(this UaTcpSecureChannel channel, BrowseRequest browseRequest)
{
	var browseResponse = await channel.BrowseAsync(browseRequest);

	var continuationPoints = browseResponse.Results.Select(br => br.ContinuationPoint).Where(cp => cp != null)
		.ToArray();
	var rds = new List<ReferenceDescription>(
		browseResponse.Results[0].References ?? new ReferenceDescription[0]);
	while (continuationPoints.Length > 0)
	{
		var browseNextRequest = new BrowseNextRequest
			{ContinuationPoints = continuationPoints, ReleaseContinuationPoints = false};
		var browseNextResponse = await channel.BrowseNextAsync(browseNextRequest);
		rds.AddRange(browseNextResponse.Results.Where(result => result.References != null)
			.SelectMany(result => result.References));
		continuationPoints = browseNextResponse.Results.Select(br => br.ContinuationPoint)
			.Where(cp => cp != null).ToArray();
	}

	return rds;
}
```
```csharp
private async Task _BrowseDataBlocks(UaTcpSessionChannel channel, ReferenceDescription referenceDescription)
{
	var staticResponse = await channel.BrowseStaticAsync(referenceDescription);

	foreach (var reference in staticResponse)
	{
		if (reference.IsIcon())
		{
			continue;
		}

		if (reference.IsOpcStructOrDatablock(out string structName))
		{
			// Here is an automatism to create subscriptions etc. automatically.
		}
		else
		{

			await _BrowseDataBlocks(channel, reference);
		}
	}
}
```
```csharp
private static readonly Regex _IsOpcStructOrDatablockRegex = new Regex("^[V,O]T_\"(HTOPC_.*)\"$"); // ^[V,O]T_(HTOPC_\w*)$
public static bool IsOpcStructOrDatablock(this ReferenceDescription description, out string name)
{
	return description._RegexMatch(_IsOpcStructOrDatablockRegex, out name);
}
private static bool _RegexMatch(this ReferenceDescription description, Regex regex, out string name)
{
	var identifier = description.TypeDefinition.NodeId.Identifier.ToString();
	var match = regex.Match(identifier);

	name = match.Groups[1].Value;

	return match.Success;
}
```

dojo90, 2019-12-19:
I have rewritten my code and achieved an amazing speed increase from initially `22s` to about `4-5s`
 🎊🎉! 

1. Filtering all unnecessary `ReferenceDescription` which cannot contain any `udt/struct` or `function block`
 - increase from `22s` to `7s`
2. I'm collecting all `ReferenceDescriptions` before using the `UaTcpSessionChannel` and use one request to get them all
- increase from `7s` to `4-5s`

---

So actually I'm happy to got this performance boost, but the main problem still exists. If using more than one `UaTcpSessionChannel` to discover the server, it will be slower than if you only use one connection at the same time.

awcullen, 2019-12-31:
I am happy that your benchmarks lead you to good performance.  Thanks, @dojo90 and @quinmars and Happy New Year.

closed 2019-12-31 by awcullen

dojo90, 2020-01-10:
Hello @awcullen,
I wish you a happy new year, too!

Unfortunately the issue is not fixed yet. I only found a workaorund to reduce my loading time (but I have a very small project here, for bigger projects this will look completely different).

Can you please reproduce the scenario to reproduce the bug? The point is that multiple parallel connections are always slower than 1 connection. If I have enough resources available on the server, then this should not be the case.

Will you please reopen the issue? Thanks!

dojo90, 2020-02-13:
For anyone else having this problem:

Add as many `NodeId` in your `BrowseRequest` as possible, to reduce the amount of total requests. So I'm able to browse more than 1000 `NodeIds` in less than 3 seconds.. 👍