convertersystems/opc-ua-client #129: Structure Binary Encoding Id from configuration

krzyfre, 2020-01-15:
I'm connecting to two different vendors PLC's (Omron and B&R). I have exactly the same structures in both PLC's. Because of the way, that OPC is implemented in both of them the BinaryEncodingId is different for each one of them.
Is there a different way of passing BinaryEncodingId to a class that inherits from Structure other than through annotations?

What I did so far is I implemented a class:

```
public class Test1 : Structure
{
        public string test1 { get; set; }
        public string test2 { get; set; }
}

public override void Encode(IEncoder encoder)
{
        encoder.WriteString("test1 ", this.test1 );
        encoder.WriteString("test2 ", this.test2 );
}

public override void Decode(IDecoder decoder)
{
        this.test1 = decoder.ReadString("test1 ");
        this.test2 = decoder.ReadString("test2");
}

```
And then I have two classes with annotations:

        [BinaryEncodingId("ns:4; i=5001")]
        public class OmronTest1:Test1  {}

        [BinaryEncodingId("ns:6; i=100001")]
        public class BRTest1:Test1  {}

But it gets more complicated, when I have more nested structures inside.

quinmars, 2020-01-20:
That's an interesting question, but I fear it is currently not possible. There a several options how that could be solved.

 1. Allow more than one attribute per class so you could write:
```csharp
[BinaryEncodingId("ns:4; i=5001")]
[BinaryEncodingId("ns:6; i=100001")]
public class Test1 : Structure
{
...
}
```
2. Change the `ÀdditionalTypes` parameter of the tcp channel to be of the type `KeyValuePair<ExpandedNodeId, Type>` so that the you can pass the expanded node explicitly.

3. With PR #131 a custom encoding map could maybe inject via the constructor.

Option 1 would probably the easiest way, since it could be done without break the existing API. Option 3 could become more flexible on the long run, but I think it will take some time until there will be a 3.0 release.

quinmars, 2020-01-20:
Oh, option 1 cannot work, because writing a `Test1` class would be ambiguous. We need a one to one mapping of type to node Id.

krzyfre, 2020-01-21:
Would it be possible to add an overlad for ```AdditionalTypes```, which takes ```KeyValuePair<ExpandedNodeId, Type>``` to keep backwards compatibility. I know it would impact quite a few constructors of classes, but maybe it would be the best option?

awcullen, 2020-01-21:
Krzyfre,  if you don't wish to create so many types(with attributes), you can still read and write DataValues.  You will just receive an ExtensionObject as the value.  The ExtensionObject is just the encoded byte array and the encoding id.  
If, for instance, you read a DataValue from Omron and wish to write to B&R, you can create a new ExtensionObject with the byte array you received from Omron and use the encoding id of B&R.

closed 2020-05-07 by awcullen