convertersystems/opc-ua-client #130: Barebones example using Forms instead of XAML

tangelo-IEM, 2020-01-16:
Good evening,

First of all, I'm very green on C#, I usually only program embedded in plain C, so using C# is being a wild ride..

I found your library that is way better to understand than the OPC UA Foundation examples, can't make heads or tails of all that code.

I have your console example running, and I got it to read a couple vars using the MonitoredItemCreateRequest, but I couldn't get MonitoredItemCreateResult to work so I used the simpler writeRequest to write to a value.

Then I tried to make a GUI using forms, but it all went wrong due to everything being async, I can't wrap my head on how to even have a connect working, 

`private static async Task Connect(string hostname, string port)
        {
            var discoveryURL = $"opc.tcp://{hostname}:{port}";

            var appDescription = new ApplicationDescription()
            {
                ApplicationName = "IEM13019_HMI",
                ApplicationUri = $"urn:{System.Net.Dns.GetHostName()}:IEM13019_HMI",
                ApplicationType = ApplicationType.Client,
            };

            //Create a certificate store
            var certificateStore = new DirectoryStore(
               Path.Combine(Environment.GetFolderPath(Environment.SpecialFolder.LocalApplicationData), "Workstation.ConsoleApp", "pki"));

            //Create a session with your server
            var channel = new UaTcpSessionChannel(
                appDescription,
                certificateStore,
                ShowSignInDialog,
                discoveryURL);

            try
            {
                await channel.OpenAsync();

            }
            catch (ServiceResultException ex)
            {
                if ((uint)ex.HResult == StatusCodes.BadSecurityChecksFailed)
                {
                    Console.WriteLine("Error connecting to endpoint");
                }

                await channel.AbortAsync();
                throw;
            }
        }`

If I put that being called on the start of the form without the await it sort of kinda works, but I cant get any value out of it because I can't reference a text box inside the connect function and VS complains about not having an await before the call.
If I put the await in there, it says that  the method is not async...

> CS4033-The 'await' operator can only be used within an async method. Consider marking this method with the 'async' modifier and changing its return type to 'Task'.

`public frmMain()
        {          
            InitializeComponent();
            timer_time.Start();
            Connect("localhost", "49320");
            //TestAsync();
        }`

Could you or maybe @basprins provide a barebones example using Forms?
Or guide me on how to use the XAML based way, it kinda seems like the code is cleaner, but I have no idea where to even start doing it that way.

Best regards.

awcullen, 2020-01-17:
To call an async function from frmMain() you can append .Wait(), as in:
Connect("localhost", "49320").Wait()

In your case, you may wish to return the channel you just opened.  You can change your function sig to:
private static async Task<UaTcpSessionChannel> Connect(string hostname, string port)

and add this line after OpenAsync():
return channel;

tangelo-IEM, 2020-01-21:
Good morning @awcullen , thanks for your quick reply.
For now I had to halt my focus on OPC-UA, but when I get the time to get back to it I will start with your suggestions and work from there.
I think for now that its best to close this "issue".

I will be back when I have it working, best regards,
Tiago.

closed 2020-01-21 by tangelo-IEM