convertersystems/opc-ua-client #133: The bytes [E0] in index 0 cannot be translated to Unicode from the specified code page.

dojo90, 2020-02-05:
I get the following `exception` on browsing a `Siemens 1200 v4.4`

```console
   bei Workstation.ServiceModel.Ua.Channels.CommunicationObject.ThrowIfClosedOrNotOpening()
   bei Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel.<RequestAsync>d__119.MoveNext()
   bei System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   bei System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   bei Workstation.ServiceModel.Ua.ViewServiceSet.<BrowseAsync>d__0.MoveNext()
   bei System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   bei System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   bei System.Runtime.CompilerServices.TaskAwaiter`1.GetResult()
```

> Die Bytes [E0] im Index 0 können nicht von der angegebenen Codepage in Unicode übersetzt werden.
> The bytes [E0] in index 0 cannot be translated to Unicode from the specified code page.

```csharp
browseRequest = new BrowseRequest
{
	NodesToBrowse = new BrowseDescription[]
	{
		new BrowseDescription
		{
			NodeId = NodeId.Parse(ObjectIds.RootFolder),
			BrowseDirection = BrowseDirection.Forward,
			ReferenceTypeId = NodeId.Parse(ReferenceTypeIds.HierarchicalReferences),
			NodeClassMask = (uint) NodeClass.Variable | (uint) NodeClass.Object |
							(uint) NodeClass.Method,
			IncludeSubtypes = true, ResultMask = (uint) BrowseResultMask.All
		}
	},
};

BrowseResponse browseResponse;
try
{
	browseResponse = await channel.BrowseAsync(browseRequest);

}
catch (Exception ex)
{
	return;
}
```

[image]

quinmars, 2020-02-08:
I think it would be useful if you could post the full stacktrace. The exception message sounds if it happens some where in the binary decoder.

awcullen, 2020-02-09:
I changed the decoder to handle errors such as this one. Let us know if v2.5.1 avoids the exception.

quinmars, 2020-02-09:
I'm pretty sure your commit will do the trick. I wonder why the Siemens server returns an invalid UTF-8 string, why does it even return a none 7-bit ASCII string? Maybe variable comments?

dojo90, 2020-02-10:
First of all, the update has fixed the bug / no more exception was thrown 👍

**But which was very strange though.**
When I did the test again this morning (before the update) and set a breakpoint before `SubscriptionDiagnosticsArray`, no error occurred after manual step over! When I removed the breakpoint the error occurred again immediately. 🤨

closed 2020-02-15 by awcullen