convertersystems/opc-ua-client #134: User identity token is valid but the server has rejected it

aarongdl, 2020-02-09:
Hello;

I'm looking to write a Xamarin application to connect to our local OPC Server (KepWare) to read and write a few tag values.

I'm able to compile and build the application using the public opc endpoint; but I'm not having any success connecting to my local install of Kepware. I've removed security on the URL endpoints and enabled them both (I would like to use security if I could figure out how to generate a certificate and then import it) but even with no security I am getting the server is rejecting the connection.

I've used both the IP address along with the hostname of the machine running kepware.

awcullen, 2020-02-09:
The title of your post indicates that the Kepware server is rejecting the User Identity. The user identity can be null (Anonymous) or a UserName (and password).  Try checking Kepware's UA Configuration settings to be sure you have enabled anonymous access.

One more thing to check: if you are running your Xamarin app in Android Emulator, and you wish to connect to a server running on the localhost, you will need to use 'opc.tcp://10.0.2.2:49320' as the Url.  More info here:  https://stackoverflow.com/questions/9808560/why-do-we-use-10-0-2-2-to-connect-to-local-web-server-instead-of-using-computer

awcullen, 2020-02-09:
I should correct the above post.  'Allow Anonymous login' is found in project properties.
[image]

aarongdl, 2020-02-09:
Hi Andrew,

Thank you so much! It looks like I was able to connect.
How can I go about generating a certificate for a specific android device and use that?

closed 2020-02-10 by aarongdl