convertersystems/opc-ua-client #138: Reconnecting to existing session with new secure channel

PatrickAulbach, 2020-03-06:
We have an issue regarding reconnecting to an existing session. 
In the opc ua standard, reconnecting is described as:
> "To do this the Client shall re-establish the connection by creating a new SecureChannel and activating the Session with the Service ActivateSession. This assigns the new SecureChannel to the existing Session and allows the Client to reuse the Session and Subscriptions in the Server."

We wanted to create a reconnect method based on this standard, but we don't a see a way to reactivate a existing session with a new channel.
Is there a way to assign the session id from a faulted channel to a new channel?

PatrickAulbach, 2020-03-12:
I made some progress in the meantime. I forked the project and created a new constructor in UaTcpSessionChannel in which one can pass an Object which holds the necessary variabled to reconnect. 
This is working at the moment.

awcullen, 2020-03-12:
Excellent.  I will look to see if I can incorporate your work into the main branch.

closed 2020-03-12 by awcullen