convertersystems/opc-ua-client #139: UnobservedTaskException after closing UaTcpSessionChannel

azborgonovo, 2020-03-17:
Observed in version 2.4.0

**StackTrace**

   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel.<RequestAsync>d__119.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Workstation.ServiceModel.Ua.SubscriptionServiceSet.<PublishAsync>d__3.MoveNext()

**Analysis**

UaTcpSessionChannel.OpenAsync method starts a new Task which is not observed:
`this.stateMachineTask = Task.Run(() => this.StateMachineAsync(this.stateMachineCts.Token));`
This would not be a problem if the method this.StateMachineAsync ensures that no exception is thrown.

However, I believe that the task composition in the PublishAsync method is creating an unobserved task exception.
The following code (UaTcpSessionChannel.PublishAsync, line 787), for example, is one way I think would solve this error:

From:
```
var publishResponse = await this.PublishAsync(publishRequest).WithCancellation(token);

// post to linked data flow blocks and subscriptions.
this.publishResponses.Post(publishResponse);
```

To:
```
var publishTask = this.PublishAsync(publishRequest);
var cancellationTask = publishTask.WithCancellation(token);

await Task.WhenAll(publishTask, cancellationTask).ConfigureAwait(false);

// post to linked data flow blocks and subscriptions.
var publishResponse = await publishTask;
this.publishResponses.Post(publishResponse);
```

I can provide extra information if needed.

Thanks in advance.

awcullen, 2020-03-18:
Yes, I see the problem.  Thank you very much for finding the solution.

azborgonovo, 2020-03-25:
Thanks for the fast reply Andrew!
Do you have an idea of which version/when will you be able to fix it?

awcullen, 2020-03-26:
I have uploaded v2.5.2 to NuGet and would love to hear if it fixes this issue.

azborgonovo, 2020-04-02:
Thanks! I will test it as soon as possible and the I give you feedback.

closed 2020-05-07 by awcullen

azborgonovo, 2020-06-17:
Better late than never: it worked, fix solved the issue!
Thank you!