convertersystems/opc-ua-client #140: Alarm and Condition with  2.1.1

rebj1011, 2020-03-17:
Hey andrew
I'm a bit lost :(.
I'm trying to program an Client which can read all alarms from an OPC-UA Server. 
The NodeId is the right one because with UAExpert and the .Net-Legacy-Sample Client I get what I want.
the following code snippet is what i tryed but i'm a bit stuck. I'm not sure what to put in the brackets after the token call.

`
Console.WriteLine("Step 5 - Add items to the subscription.");
   var alarmitemsToCreate = new MonitoredItemCreateRequest[]
                {   new MonitoredItemCreateRequest { ItemToMonitor = new ReadValueId { NodeId = NodeId.Parse("..."), AttributeId = AttributeIds.EventNotifier }, MonitoringMode = MonitoringMode.Reporting, RequestedParameters = new MonitoringParameters { ClientHandle = 00001, SamplingInterval = -1, QueueSize = 0, DiscardOldest = true } },  };

                var itemsRequest = new CreateMonitoredItemsRequest
                {
                    SubscriptionId = id,
                    ItemsToCreate = alarmitemsToCreate,
                };

                var itemsResponse = await Channel.CreateMonitoredItemsAsync(itemsRequest);

               Console.WriteLine("Step 6 - Subscribe to PublishResponse stream.");
                token = Channel.Where(pr => pr.SubscriptionId == id).Subscribe(pr =>
                {???}
 });

            }
                 `

can you help me ? Is my attempt right or is it already wrong?

awcullen, 2020-03-18:
Hi,
Take a look at an example of Event Subscription at the link below.  Let me know what questions you have, as the sample has few comments.  Your questions help me make the examples better.

 https://github.com/convertersystems/OpcUaFeatureTests/blob/f358886d6a9b9873801dd968a34154b51c5f311a/OpcUaFeatureTests/SubscriptionTests.cs#L125

rebj1011, 2020-03-18:
I tryed your sample but i never get in your foreach-loop

 `
 foreach (var enl in enls)
                            {
                                foreach (var efl in enl.Events)
                                {
                                    var ev = EventHelper.Deserialize<BaseEvent>(efl.EventFields);
                                    Console.WriteLine($"time: {ev.Time}, src: {ev.SourceName}, msg: {ev.Message}, sev: {ev.Severity}");
                                }
                            }
`
Do i have to add something?

rebj1011, 2020-03-18:
or do i have to change the Filter
`Filter= new EventFilter{SelectClauses= EventHelper.GetSelectClauses<BaseEvent>()`
in somekind of alarmfilter or something like that?

rebj1011, 2020-03-18:
somewhere else I found something like this
`            tmpEventItem.Filter.SelectClauses.Add(new QualifiedName(BrowseNames.AckedState));
            tmpEventItem.Filter.SelectClauses.Add(new QualifiedName(BrowseNames.Severity));
            tmpEventItem.Filter.SelectClauses.Add(new QualifiedName(BrowseNames.SourceName));`
as filter definition

awcullen, 2020-03-18:
So this EventFilter selects the fields of a BaseEvent, which is the base class for all events and alarms.
Filter= new EventFilter{SelectClauses= EventHelper.GetSelectClauses<BaseEvent>()

You should be getting every event that triggers.

rebj1011, 2020-03-18:
Okay i understand that.
But how is it possible that i couldn't get any event :P?
In my UAExpert i get some.
[image]

Do you have an idea where my mistake is?

awcullen, 2020-03-18:
Would you be able to share the code you have?

rebj1011, 2020-03-18:
I implemened 100% of your example i'm not using a AnonymousIdentity i use a User with all rights to read and write. All other function like normal subscription, write and read work like a charm. Only the EventSubscription doesn't give me the results.

`  var loggerFactory = new LoggerFactory();

            var discoveryUrl = _serverAddress.ToString(); 
                                                          
            Console.WriteLine("Step 1 - Describe this app.");
            var appDescription = new ApplicationDescription()
            {
                ApplicationName = "...",
                ApplicationUri = $"urn:{System.Net.Dns.GetHostName()}:...",
                ApplicationType = ApplicationType.Client,
            };

            Console.WriteLine("Step 2 - Create a certificate store.");
            var certificateStore = new DirectoryStore(
                Path.Combine(Environment.GetFolderPath(Environment.SpecialFolder.LocalApplicationData), "ClientServerApp", "pki"));
            try {
                Console.WriteLine("Step 3 - Create a session with your server.");
                Channel = new UaTcpSessionChannel(
                appDescription,
                certificateStore,
                ShowSignInDialog,
                discoveryUrl,
                loggerFactory: loggerFactory);

            try
            {
                // try opening a session and reading a few nodes.
                await Channel.OpenAsync();

                // build a CreateSubscriptionRequest. See 'OPC UA Spec Part 4' paragraph 5.13.2
                var req = new CreateSubscriptionRequest
                {
                    RequestedPublishingInterval = 500.0, // intervals are in milliseconds
                    RequestedMaxKeepAliveCount = 30,
                    RequestedLifetimeCount = 30 * 3,
                    PublishingEnabled = true,
                };

                var res = await Channel.CreateSubscriptionAsync(req);

                // the result will return the server's subscription id. You will needs this to 
                // add monitored items.
                var id = res.SubscriptionId;
                Console.WriteLine($"Created subscription '{id}'.");

                // build a CreateMonitoredItemsRequest. See 'OPC UA Spec Part 4' paragraph 5.12.2
                var req2 = new CreateMonitoredItemsRequest
                {
                    SubscriptionId = id,
                    TimestampsToReturn = TimestampsToReturn.Both,
                    ItemsToCreate = new MonitoredItemCreateRequest[]
                    {
                    new MonitoredItemCreateRequest
                    {
                        ItemToMonitor= new ReadValueId{ AttributeId= AttributeIds.EventNotifier, NodeId= NodeId.Parse("...")},
                        MonitoringMode= MonitoringMode.Reporting,
                        // specify a unique ClientHandle. The ClientHandle is returned in the PublishResponse
                        RequestedParameters= new MonitoringParameters{ ClientHandle= 42, SamplingInterval=-1.0, QueueSize=1000, DiscardOldest=true,
                            // events require an EventFilter with a SelectClause (a list of fields to receive)
                            Filter= new EventFilter{SelectClauses= EventHelper.GetSelectClauses<BaseEvent>() } },
                    },
                    },
                };

                
                var res2 = await Channel.CreateMonitoredItemsAsync(req2);

                Console.WriteLine("\nSubscribe to PublishResponse stream.");

                // when the session is open, the client sends a stream of PublishRequests to the server.
                // You can subscribe to all the PublishResponses -or- subscribe to the responses from 
                // a single subscription.
                var token = Channel
                    // receive responses for the subscription we just created
                    .Where(pr => pr.SubscriptionId == id)
                    // subscribe with an 'OnNext' function, and an 'OnError' function
                    .Subscribe(
                        pr => 
                        {
                            
                            // loop thru all the event notifications and write them out.
                            var enls = pr.NotificationMessage.NotificationData.OfType<EventNotificationList>();
                            foreach (var enl in enls)
                            {
                                foreach (var efl in enl.Events)
                                {
                                    var ev = EventHelper.Deserialize<BaseEvent>(efl.EventFields);
                                    Console.WriteLine($"time: {ev.Time}, src: {ev.SourceName}, msg: {ev.Message}, sev: {ev.Severity}");
                                }
                            }

                            
                    
                        },
                        ex => Console.WriteLine("Exception in publish response handler: {0}", ex.GetBaseException().Message)
                   );

            

                    while (true)
                {

                  
                    await Task.Delay(500);
                }
                //}

                Console.WriteLine($"\nClosing session '{Channel.SessionId}'.");
                await Channel.CloseAsync();
            }
            catch (Exception ex)
            {
                await Channel.AbortAsync();
                Console.WriteLine(ex.Message);
            }

        }

`

rebj1011, 2020-03-18:
After step three there is an catch and a }

awcullen, 2020-03-18:
I subscribed to nodeid "i=2253". (which is the 'server' node)

At first I didn't get any events.
I triggered an event using UAExpert.  In my case, the server is UaCppServer and you can trigger an event by turning bool tag "ns=2;s=Demo.Events.Trigger_SampleEvent"
The event was received.  

An event subscription does not receive any event that happened in the past.

rebj1011, 2020-03-18:
Okay but shouldn't be alarms also be recognized as events? Or do i misunderstand something?

awcullen, 2020-03-18:
Yes,  alarms are events and if you trigger a new alarm it should be received by the subscription.

When I set the value to true of the node "ns=2;s=AlarmsWithNodes.OffAlarmTrigger" I get a new alarm and a new event.

rebj1011, 2020-03-18:
I'm a bit confused i guess i download the UaCppServer and try to trigger some events maybe this helps. 
i keep you uptodate.

rebj1011, 2020-04-02:
Hi andrew, sorry for my delayed update.
With the UACppServer everything works fine, but with the server on the siemens sinumerik i couldn't read an single event. Is it possible to get an version of the UAClient lib for .Net framework 4.6.0 so i can debugg it with my visual studio 2015 and may find my problem?

awcullen, 2020-04-02:
Microsoft tells me:

> If you only need to consume .NET Standard 2.0 libraries in your projects, you can also do that in Visual Studio 2015. However, you need NuGet client 3.6 or higher installed. You can download the NuGet client for Visual Studio 2015 from the NuGet downloads 

Also you would need to target .Net framework 4.6.1 , I think.  
Would that be a problem?

rebj1011, 2020-04-03:
I updated my NuGet and i target  .Net framework 4.6.1, but still i can't load your lib. Get this error message
"at least one project in the project portfolio was not carried out properly".
And as output i get
"The MSBuild XML namespace must be the project's default XML namespace. If the project was created in MSBuild 2003 format, add xmlns="http://schemas.microsoft.com/developer/msbuild/2003" to the <Project> element. If the project was created in the old 1.0 or 1.2 format, convert it to the MSBuild 2003 format. "

rebj1011, 2020-04-22:
I found the problem. The "Deserialize function" couldn't convert the comming note.

 var token = Channel
                    // receive responses for the subscription we just created
                    .Where(pr => pr.SubscriptionId == id)
                    // subscribe with an 'OnNext' function, and an 'OnError' function
                    .Subscribe(
                        pr =>
                        {
                            try
                            {                             
                            // loop thru all the event notifications and write them out.
                            var enls = r.NotificationMessage.NotificationData.OfType<EventNotificationList>();
                                foreach (var enl in enls)
                                {
                                    foreach (var efl in enl.Events)
                                    { 
                                        var ev = EventHelper.Deserialize<BaseEvent>(efl.EventFields);
                                    }
                                }                               
                        },

Thanks for your Help.

closed 2020-04-22 by rebj1011

harishLinkEZ, 2021-03-30:
Hi rebj1011,
I had been stuck on the same issue from a week around, no idea how to proceed. can you  please help?

I have tried same with siemens sinumeric,  but could not able to read any event or alarms, what is the node it you had subscribed to ?
Can you share the code snippet of the alarm or event subscription ?

This is the code below: I am able to read nodes properly , could create subscription , but no data whatsoever on the events & alarms part. Can you please help.

`
 public static async Task CreateEventSubscription()
{
// describe this client application.
var discoveryUrl = $"opc.tcp://" + "192.168.1.1 + ":" + 4840;

        Console.WriteLine("Step 1 - Describe this app.");
        var appDescription = new ApplicationDescription()
        {
            ApplicationName = "OPCWorkstation",
            ApplicationUri = $"urn:{System.Net.Dns.GetHostName()}:OPCWorkstation",
            ApplicationType = ApplicationType.Client,
        };

        Console.WriteLine("Step 2 - Create a certificate store.");
        var certificateStore = new DirectoryStore(
            Path.Combine(Environment.GetFolderPath(Environment.SpecialFolder.LocalApplicationData), "OPCWorkstation.ConsoleApp", "pki"));//Workstation.UaClient.

        Console.WriteLine("Step 3 - Create a session with your server.");
        var channel = new UaTcpSessionChannel(
            appDescription,
            certificateStore,
            new UserNameIdentity("OpcUaClient", "OpcUaClient"),
            discoveryUrl,
            SecurityPolicyUris.Basic128Rsa15);

        try
        {
            await channel.OpenAsync();

            Console.WriteLine($"  Opened channel with endpoint '{channel.RemoteEndpoint.EndpointUrl}'.");
            Console.WriteLine($"  SecurityPolicyUri: '{channel.RemoteEndpoint.SecurityPolicyUri}'.");
            Console.WriteLine($"  SecurityMode: '{channel.RemoteEndpoint.SecurityMode}'.");
            Console.WriteLine($"  UserIdentity: '{channel.UserIdentity}'.");
            Console.WriteLine($"Opened session with endpoint '{channel.RemoteEndpoint.EndpointUrl}'.");
            Console.WriteLine($"SecurityPolicy: '{channel.RemoteEndpoint.SecurityPolicyUri}'.");
            Console.WriteLine($"SecurityMode: '{channel.RemoteEndpoint.SecurityMode}'.");
            Console.WriteLine($"UserIdentityToken: '{channel.UserIdentity}'.");

            // build a CreateSubscriptionRequest. See 'OPC UA Spec Part 4' paragraph 5.13.2
            var req = new CreateSubscriptionRequest
            {
                RequestedPublishingInterval = 500.0, // intervals are in milliseconds
                RequestedMaxKeepAliveCount = 30,
                RequestedLifetimeCount = 30 * 3,
                PublishingEnabled = true,
            };
            var res = await channel.CreateSubscriptionAsync(req);

            // the result will return the server's subscription id. You will needs this to 
            // add monitored items.
            var id = res.SubscriptionId;
            Console.WriteLine($"Created subscription '{id}'.");

            NodeId eventNodeId;
            NodeId.TryParse("ns=2;s=Sinumeric", out eventNodeId);
            
            // build a CreateMonitoredItemsRequest. See 'OPC UA Spec Part 4' paragraph 5.12.2
            var req2 = new CreateMonitoredItemsRequest
            {
                SubscriptionId = id,
                TimestampsToReturn = TimestampsToReturn.Both,
[... 36 lines cut ...]
                        {
                            foreach (var efl in enl.Events)
                            {
                                var ev = EventHelper.Deserialize<BaseEvent>(efl.EventFields);
                                Console.WriteLine($"time: {ev.Time}, src: {ev.SourceName}, msg: {ev.Message}, sev: {ev.Severity}");
                            }
                        }
                    },
                    ex => Console.WriteLine("Exception in publish response handler: {0}", ex.GetBaseException().Message)
               );
            while(true)
            {
                Task.Delay(500);
            }
            await Task.Delay(500);
        }
        catch (Exception ex)
        {
            await channel.AbortAsync();
            Console.WriteLine(ex.Message);
        }
    }
  
}
`

awcullen, 2021-03-31:
Hi HarishLinkEZ,
@rebj1011 seems to say that there was an error in the  EventHelper.Deserialize function below:

```
        foreach (var efl in enl.Events)
        {
            var ev = EventHelper.Deserialize<BaseEvent>(efl.EventFields);
            Console.WriteLine($"time: {ev.Time}, src: {ev.SourceName}, msg: {ev.Message}, sev: {ev.Severity}");
        }
```

Instead you can try:

```
        foreach (var efl in enl.Events)
        {
            Console.WriteLine(string.Join(", ", efl.EventFields.Select(e => e.GetValue())));
        }

```

harishLinkEZ, 2021-03-31:
Dear Andrew,
Thanks for the code snippet. I got it working. My problem was improper node name & user access rights. My code wasn't even getting into the first loop.
For anyone working with Siemens sinumeric: here is a piece of guidance.

1) Node name for events is "ns=2;s=Sinumerik" . NOT "ns=2;s=Sinumeric"
2) "ns=2;s=Sinumerik" is the only node present with events
```
NodeId eventNodeId;
NodeId.TryParse("ns=2;s=Sinumerik", out eventNodeId);
```
3) You have to give access to the user "ns=2;s=Sinumerik/Methods/GiveUserAccess" using your username & "SinuReadAll", Find the details how to give user access to events in following manual:
https://cache.industry.siemens.com/dl/files/643/109772643/att_1001368/v1/840Dsl_828D_OPCUA_config_man_0819_en-US.pdf
under alarms&events section.

Code example:
```
 var discoveryUrl = $"opc.tcp://" + <machineIp> + <tcpport: usually 4840 for sinumerik>
            UaTcpSessionChannel Channel;
            Console.WriteLine("Step 1 - Describe this app.");
            var appDescription = new ApplicationDescription()
            {
                ApplicationName = "...",
                ApplicationUri = $"urn:{System.Net.Dns.GetHostName()}:...",
                ApplicationType = ApplicationType.Client,
            };

            Console.WriteLine("Step 2 - Create a certificate store.");
            var certificateStore = new DirectoryStore(
                Path.Combine(Environment.GetFolderPath(Environment.SpecialFolder.LocalApplicationData), "ClientServerApp", "pki"));
           
            Console.WriteLine("Step 3 - Create a session with your server.");
            Channel = new UaTcpSessionChannel(
            appDescription,
            certificateStore,
            new UserNameIdentity("OpcUaClient", "OpcUaClient"),
            discoveryUrl);

            try
            {
                // try opening a session and reading a few nodes.
                await Channel.OpenAsync();

                // build a CreateSubscriptionRequest. See 'OPC UA Spec Part 4' paragraph 5.13.2
                var req = new CreateSubscriptionRequest
                {
                    RequestedPublishingInterval = 500.0, // intervals are in milliseconds
                    RequestedMaxKeepAliveCount = 30,
                    RequestedLifetimeCount = 30 * 3,
                    PublishingEnabled = true,
                };

                var request = new CallRequest
                {
                    MethodsToCall = new[] {
                    new CallMethodRequest
                    {
                        //ObjectId = NodeId.Parse("s=Sinumeric/Methods"),
                        MethodId = NodeId.Parse("ns=2;s=Sinumerik/Methods/GiveUserAccess"),
                        InputArguments = new [] {"OpcUaClient", "SinuReadAll"}.ToVariantArray()
                    }
                }
                };
                var response = await Channel.CallAsync(request);
}
catch
{
}
```

harishLinkEZ, 2021-03-31:
@awcullen I really appreciate your work and I want to contribute back to the repository by adding basic samples for using siemens/sinumerik machines. This could help first timers save lot of bucks.

quinmars, 2021-04-02:
I like the idea. It'd be greate to have a collection of very basic examples/snippets. There are samples for the MVVM part in the 
`opc-ua-samples` respository. And you find many very good examples in the `OpcUaFeatureTests` respository. The latter are impressively good. There are many educational comments and references to the standard. Unfortunately they are camouflaged as feature tests. In my vision we would have an directory (be it in the main repository or in another one), which contains single file, single pupose console application for each service. The structure would be like

```
examples\Connect\Program.cs
examples\ConnectWithSecurity\Program.cs
examples\ConnectWithUserNameAndPassword\Program.cs
examples\ReadValue\Program.cs
examples\ReadValues\Program.cs
examples\ReadAttributes\Program.cs
...
examples\SinumerikAlarms\Program.cs
```

And your working example could be one of it.