convertersystems/opc-ua-client #142: Simple connect with WPF app

AKFrick, 2020-03-24:
Attempt to connect to S7-1500.

```
public async Task UaClient()
        {
            // describe this client application.
            var clientDescription = new ApplicationDescription
            {
                ApplicationName = "Workstation.UaClient.FeatureTests",
                ApplicationUri = $"urn:{System.Net.Dns.GetHostName()}:Workstation.UaClient.FeatureTests",
                ApplicationType = ApplicationType.Client
            };

            // create a 'UaTcpSessionChannel', a client-side channel that opens a 'session' with the server.
            var channel = new UaTcpSessionChannel(
                clientDescription,
                null, // no x509 certificates
                new AnonymousIdentity(), // no user identity
                "opc.tcp://192.168.10.2:4840", // the public endpoint of a server at opcua.rocks.
                SecurityPolicyUris.None); // no encryption
            try
            {
                // try opening a session and reading a few nodes.
                await channel.OpenAsync();
                Log.GetLog().logThis($"Opened session with endpoint '{channel.RemoteEndpoint.EndpointUrl}'.");
                Log.GetLog().logThis($"SecurityPolicy: '{channel.RemoteEndpoint.SecurityPolicyUri}'.");
                Log.GetLog().logThis($"SecurityMode: '{channel.RemoteEndpoint.SecurityMode}'.");
                Log.GetLog().logThis($"UserIdentityToken: '{channel.UserIdentity}'.");
                var readRequest = new ReadRequest
                {
                    // set the NodesToRead to an array of ReadValueIds.
                    NodesToRead = new[] {
                    // construct a ReadValueId from a NodeId and AttributeId.
                    new ReadValueId {
                        // you can parse the nodeId from a string.
                        // e.g. NodeId.Parse("ns=2;s=Demo.Static.Scalar.Double")
                        NodeId = NodeId.Parse(VariableIds.Server_ServerStatus),
                        // variable class nodes have a Value attribute.
                        AttributeId = AttributeIds.Value
                    },
                    new ReadValueId
                    {
                        NodeId = new NodeId("\"OPCUA\".\"MyInt\"", 3),
                        AttributeId = AttributeIds.Value
                    }
                }
                };
                var readResult = await channel.ReadAsync(readRequest);
                var MyInt = readResult.Results[1];
                Log.GetLog().logThis($" OPCUA.MyInt: { MyInt.GetValueOrDefault<Int16>()}");
                await channel.CloseAsync();
            }
            catch (Exception ex)
            {
                await channel.AbortAsync();
                Log.GetLog().logThis(ex.Message);
            }
        }
```
For Console App it works fine but from XAML app, I got exception: "System.ObjectDisposdException in System.dll ("Cannot access a disposed object")
I call this method just like 
```
        private void Button_Click(object sender, RoutedEventArgs e)
        {
            UaClient().Wait();
        }
```
Could you give me some explanations what am I doing wrong?

awcullen, 2020-03-24:
Hi,  This is good question.  Wpf and Console have different sync contexts.  The UaClient function you wrote has a number of async functions that need to do some network activities and then return to the sync context of the caller. You have two choices.
1)  Call your UaClient() function as a task on the thread pool.
```
        private void Button_Click(object sender, System.Windows.RoutedEventArgs e)
        {
            Task.Run(()=>UaClient());
        }
```
2)  Add .ConfigureAwait(false) to all the async calls in your function.

```
        private void Button_Click(object sender, RoutedEventArgs e)
        {
            UaClient().Wait();
        }

        public async Task UaClient()
        {
            var clientDescription = new ApplicationDescription
            {
                ApplicationName = "Workstation.UaClient.FeatureTests",
                ApplicationUri = $"urn:{System.Net.Dns.GetHostName()}:Workstation.UaClient.FeatureTests",
                ApplicationType = ApplicationType.Client
            };

            // create a 'UaTcpSessionChannel', a client-side channel that opens a 'session' with the server.
            var channel = new UaTcpSessionChannel(
                clientDescription,
                null, // no x509 certificates
                new AnonymousIdentity(), // no user identity
                "opc.tcp://127.0.0.1:48010", // the public endpoint of a server at opcua.rocks.
                SecurityPolicyUris.None); // no encryption
            try
            {
                // try opening a session and reading a few nodes.
                await channel.OpenAsync().ConfigureAwait(false);
                Console.WriteLine($"Opened session with endpoint '{channel.RemoteEndpoint.EndpointUrl}'.");
                Console.WriteLine($"SecurityPolicy: '{channel.RemoteEndpoint.SecurityPolicyUri}'.");
                Console.WriteLine($"SecurityMode: '{channel.RemoteEndpoint.SecurityMode}'.");
                Console.WriteLine($"UserIdentityToken: '{channel.UserIdentity}'.");
                var readRequest = new ReadRequest
                {
                    // set the NodesToRead to an array of ReadValueIds.
                    NodesToRead = new[] {
                    // construct a ReadValueId from a NodeId and AttributeId.
                    new ReadValueId {
                        // you can parse the nodeId from a string.
                        // e.g. NodeId.Parse("ns=2;s=Demo.Static.Scalar.Double")
                        NodeId = NodeId.Parse(VariableIds.Server_ServerStatus),
                        // variable class nodes have a Value attribute.
                        AttributeId = AttributeIds.Value
                    },
                    new ReadValueId
                    {
                        NodeId = new NodeId("\"OPCUA\".\"MyInt\"", 3),
                        AttributeId = AttributeIds.Value
                    }
                }
                };
                var readResult = await channel.ReadAsync(readRequest).ConfigureAwait(false);
                var MyInt = readResult.Results[1];
                Console.WriteLine($" OPCUA.MyInt: { MyInt.GetValueOrDefault<Int16>()}");
                await channel.CloseAsync().ConfigureAwait(false);
            }
            catch (Exception ex)
            {
                await channel.AbortAsync().ConfigureAwait(false);
                Console.WriteLine(ex.Message);
            }
        }
```

AKFrick, 2020-03-25:
It works! Thank you so much!

closed 2020-03-25 by AKFrick