convertersystems/opc-ua-client #143: OpenAsync() ServiceResultException

Pdesombres, 2020-04-01:
Hello,

After many tries, I decide to post a message.
When executing the OpenAsync() method, I have the exception below:
`Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel:Error: Error discovering endpoints of 'opc.tcp://myserverip:49380'. 
An internal error occurred as a result of a programming or configuration error.`

I don't understand, I managed to connect and read the data but since the server restart, I have this error.
Maybe a Windows service on the TOPServer is missing?
The certificate is in the TrustedCertificate.

I also tried the method UaTcpDiscoveryService.GetEndpointsAsync() but I have the same error.

The complete trace below
`Exception thrown: 'Workstation.ServiceModel.Ua.ServiceResultException' in System.Private.CoreLib.dll
Exception thrown: 'Workstation.ServiceModel.Ua.ServiceResultException' in System.Private.CoreLib.dll
Exception thrown: 'System.IO.IOException' in System.Net.Sockets.dll
Exception thrown: 'System.IO.IOException' in System.Private.CoreLib.dll
Exception thrown: 'Workstation.ServiceModel.Ua.ServiceResultException' in System.Private.CoreLib.dll
Exception thrown: 'Workstation.ServiceModel.Ua.ServiceResultException' in System.Private.CoreLib.dll
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel:Error: Error discovering endpoints of 'opc.tcp://myserverip:49380'. 
An internal error occurred as a result of a programming or configuration error.
Exception thrown: 'Workstation.ServiceModel.Ua.ServiceResultException' in Workstation.UaClient.dll
Exception thrown: 'Workstation.ServiceModel.Ua.ServiceResultException' in System.Private.CoreLib.dll
Exception thrown: 'Workstation.ServiceModel.Ua.ServiceResultException' in System.Private.CoreLib.dll
Exception thrown: 'Workstation.ServiceModel.Ua.ServiceResultException' in System.Private.CoreLib.dll`

Thanks for your help

awcullen, 2020-04-01:
Hi, 
Thanks for reporting this issue.  I see that you have tried GetEndpointsAsync() and are getting an IOException from the socket.  OPC UA  servers provide the 'GetEndpoints' service to all callers,  with no security or password neccessary. So this means that it isn't a certificate issue :).  This leads me to believe the TOP server is not running or is blocked.   Would you be able to check the endpoint with a different client?  I use UAExpert, available from https://www.unified-automation.com/products/development-tools/uaexpert.html

Andrew

Pdesombres, 2020-04-02:
Hello,

Thanks for your answer.
I try with UAExpert and have a BadInternalError. Discovery GetEndpoints failed.

Do you know if there are windows services or specific settings in TOPServer to allow the connection?

I'm going to get closer to the provider who did the TOPServer installation.

Regards,
Pierre

Pdesombres, 2020-04-03:
After a reboot of server, it's good. I don't know why but it's work.

Thanks

closed 2020-04-03 by Pdesombres