convertersystems/opc-ua-client #144: Own Certificate type

NikolayStek, 2020-04-09:
Why the own client certificate generated in *.crt format by default (DirectoryStore)? By the way in UAExpert own certificate generates in *.der format. Currently i have a Server which is not work with *.crt as trusted client certificate (only *.der are accepted)

awcullen, 2020-04-09:
You may find that your server has a tool for displaying certificates that it has rejected.  The tool often will allow you to accept the certificate. (Try right click..)

In other cases, the server may use file folders named "Trusted" and "Rejected". The server will first place the client's certificate in the "Rejected" folder and you need to move the cert file to the "Trusted" folder.

The important info is that the server will store the "rejected" certificate in the format that it uses, whether DER or PEM. 

If none of that works, then it is fairly easy to generate your client cert with OpenSSL.  Let me know and I can share a script.

NikolayStek, 2020-04-09:
The last variant helps. As I found Server is not user friendly and dotn't have "Rejected" folder(and any Utils). I convert current client CRT to DER and copy it to "Trusted" Server folder. Now I can connect with any security policy.
 
Thanks for the feedback!

If i found how to create DER in BouncyCastle I can create pull request if you interested

closed 2020-04-20 by awcullen