convertersystems/opc-ua-client #145: How to set EventFilter WhereClause

invinctus, 2020-04-21:
Hi,

I'm trying to add a where clause to an event subscription but I cannot figure out how to set the filter operands.  I was expecting to do something like this:

    new ContentFilter {
        Elements = new ContentFilterElement[]
        {
            new ContentFilterElement
            {
                FilterOperator = FilterOperator.Equals,
                FilterOperands = new FilterOperand[]{
                    new QualifiedName("SourceName"),
                    new Variant("CNC")
                }
            }
        }
    };

But the FilterOperand class has no properties and I cant figure out how I need to set the operands for the filter.

awcullen, 2020-04-21:
Great question.  The following should filter the subscription to receive only events  with "SourceName" Equals "CNC" 

```
WhereClause = new ContentFilter
{
    Elements = new ContentFilterElement[]
    {
        new ContentFilterElement
        {
            FilterOperator = FilterOperator.Equals,
            FilterOperands = new FilterOperand[]{
                new SimpleAttributeOperand{
                    BrowsePath= new []{ QualifiedName.Parse("SourceName") },
                    AttributeId=AttributeIds.Value
                },
                new LiteralOperand{ Value= new Variant("CNC") }
            }
        }
    }
}
```

invinctus, 2020-04-21:
Yep that does the trick, I was pretty close just missed those sub classes.

Thanks.

closed 2020-04-21 by invinctus