convertersystems/opc-ua-client #146: Hey Andrew,

rebj1011, 2020-04-22:


rebj1011, 2020-04-22:
Hey Andrew,
I'm sorry to bother you again. I have multiple threads and multiple subscriptions and at some point i couldn't get all trigger. 

Do you have any idea where I might find the problem?
Is it possible to speed up the RequestedPublishingInterval faster than 100ms?

awcullen, 2020-04-22:
Hi rebj.  
The server decides the actual PublishingInterval.  For instance, a Siemens S7-1500 has a default minimum PublishingInterval of 1000 ms. You can change it to 500 ms in TIA Portal, but no faster.

The server has a minimum SamplingInterval as well. The default is 1000 ms. You can change it to 100 ms in TIA Portal, but no faster.

Maybe you can tell me more about what you are looking to accomplish?   I can usually get good performance from one UaSessionChannel and creating multiple subscriptions.  I don't get better performance from multiple channels on multiple threads.

rebj1011, 2020-04-23:
Ok. 
In your 'CreateSubscriptionRequest' you have a 'RequestedPublishingInterval', isn't this the cycle to reade the values from the Server? If I set a number that is too small, will the server change it?
 
At the moment we are using just one channel for OPC communication and several threads to process the data.
Does each subscription need its own 'PublishResponse stream' for communication or is it best practice to use only one stream? If we use multible streams can one Stream somehow close another?

At the Moment we are using an trigger to read tones of data. For 7200 Bytes we need about 2sec. Can you tell whether it is fast or slow? Do you have an sample for reading lots of Data?

awcullen, 2020-04-23:
A Subscription has a PublishingInterval and a list of MonitoredItems.  Each MonitoredItem has a SamplingInterval and a Queue for storing DataValues until the Subscription is ready to publish.

Example:  PublishInterval: 1000ms,  SamplingInterval: 250ms,  QueueSize: 4 (or more)
Each MonitoredItem will sample the value and store any data change in it's Queue every 250 ms.
The Subscription will collect all the data changes and send them to you every 1000 ms.
Every 1000 ms, you will receive a PublishResponse from each Subscription with up to 4 data changes for every MonitoredItem.
  
"RequestedPublishingInterval" is the interval that you request to be notified of data changes. The CreateSubscriptionResponse will return a "RevisedPublishingInterval" that the server will actually use.

You can use one PublishResponse stream for all subscriptions or create multiple streams for multiple subscriptions.  Same performance.  The Action that handles the PublishResponse should execute quickly because a delay will cause the channel to delay in sending the next PublishRequest.  Eventually the channel will have timeout errors if the Action is blocked.  The timeout delay is actually the CreateSubscriptionRequest setting MaxLifetimeCount.  (MaxLifetimeCount * PublishingInterval = timeout)

Tell me more about the trigger and the 7200 bytes of data.  You should be able to ready 7200 bytes quickly.

rebj1011, 2020-04-24:
Thanks for the fast reply.
We have an Siemens 840D control unit and the corresponding OPC Server. In the PLC we set a bit that says "Hey you can read". I guess thats not the Problem.
The data structure is that we have a DB with many data of the data type Byte. They are all in bytes because a mix was slower.
We generate an list of Nodes. Each node read 100 DBW (200 Bytes). 
Our Nodes looks like : ns=2;s=/Plc/DB200.DBWx:[100]. x= Variable to map the hole data-area.

We use an ArrayRead function because it was faster than reading each node individually.
`

        public async Task<List<object>> OpcArrayRead(List<object> readValues)
        {

            ReadValueId[] NodesToReadlocal = new ReadValueId[readValues.Count];

            List<object> results = new List<object>();

            int i = 0;

            foreach (string node in readValues)
            {
                NodesToReadlocal[i] = new ReadValueId
                {
                    NodeId = NodeId.Parse("ns=2;s=/Random@" + node),
                    AttributeId = AttributeIds.Value
                };
                i++;
            }

            var task1 = Task.Run(async () =>
                {
                    try
                    {
                        var readRequest = new ReadRequest
                        {
                            // set the NodesToRead to an array of ReadValueIds.
                            NodesToRead = NodesToReadlocal,
                        };
                        // send the ReadRequest to the server.
                        var readResult = await Channel.ReadAsync(readRequest);

                        // DataValue is a class containing value, timestamps and status code.
                        // the 'Results' array returns DataValues, one for every ReadValueId.

                        foreach (DataValue ReadData in readResult.Results)
                        {
                            ushort[] valueInt2 = new ushort[100];

                            switch (ReadData.Variant.Type.ToString())
                            {
                                case "UInt16":
                                    results.Add(ReadData.GetValueOrDefault<ushort[]>());
                                    break;
                                case "UInt32":
                                    results.Add(ReadData.GetValueOrDefault<UInt32[]>());
                                    break;
                                case "Int16":
                                    var valueInt = ReadData.GetValueOrDefault<Int16>();
                                    results.Add(valueInt.ToString());
                                    break;
                                case "Boolean":
                                    var valueBool = ReadData.GetValueOrDefault<Boolean>();
                                    results.Add(valueBool.ToString());
                                    break;
                                case "String":
                                    var valueString = ReadData.GetValueOrDefault<String>();
                                    results.Add(valueString.ToString());
                                    break;
                                case "ByteString":
                                    object valueByteString = ReadData.GetValueOrDefault<System.Byte[]>();
                                    results.Add(valueByteString);
                                    break;
                                default:
                                    break;
                            }
                        }
                    }
                    catch (Exception exception)
                    {
                        App.Log.Error("OPCConsumer::OpcArrayRead  - ERROR " + "1003" + "- " +exception.ToString());
                    }
                });

            task1.GetAwaiter().GetResult();
            return results;
        }
`
This is our array reader.
After this we have a list with all the data and convert it. The conversion is not the problem.

I hope you can halp me with that.

awcullen, 2020-04-25:
If you call your method OpcArrayRead with the same list every time. You may benefit from a little known method named RegisterNodes.  The idea is that you pass in a list of nodeIds to RegisterNodesAsync and the server returns a list of nodeids that may provide faster performance for reading.  In actual use, the performance boost depends on the author of the server. 

```
        // Registers a list of NodeIds for use by OpcArrayRead.
        // Call this one time, after opening the Channel.
        public async Task<ReadValueId[]> OpcRegisterNodes(List<object> readValues)
        {
            NodeId[] nodesToRegister = new NodeId[readValues.Count];
            int i = 0;

            foreach (string node in readValues)
            {
                nodesToRegister[i] = NodeId.Parse("ns=2;s=/Random@" + node);
                i++;
            }

            var registerRequest = new RegisterNodesRequest { NodesToRegister = nodesToRegister };
            var registerResult = await Channel.RegisterNodesAsync(registerRequest);

            ReadValueId[] NodesToReadlocal = new ReadValueId[readValues.Count];
            i = 0;

            foreach (var nodeId in registerResult.RegisteredNodeIds)
            {
                NodesToReadlocal[i] = new ReadValueId
                {
                    NodeId = nodeId,
                    AttributeId = AttributeIds.Value
                };
                i++;
            }

            return NodesToReadlocal;
        }

        public async Task<List<object>> OpcArrayRead(ReadValueId[] NodesToReadlocal)
        {

            List<object> results = new List<object>();
            try
            {
                var readRequest = new ReadRequest
                {
                    // set the NodesToRead to an array of ReadValueIds.
                    NodesToRead = NodesToReadlocal,
                };
                // send the ReadRequest to the server.
                var readResult = await Channel.ReadAsync(readRequest);

                // DataValue is a class containing value, timestamps and status code.
                // the 'Results' array returns DataValues, one for every ReadValueId.

                foreach (DataValue ReadData in readResult.Results)
                {
                    ushort[] valueInt2 = new ushort[100];

                    switch (ReadData.Variant.Type.ToString())
                    {
                        case "UInt16":
                            results.Add(ReadData.GetValueOrDefault<ushort[]>());
                            break;
                        case "UInt32":
                            results.Add(ReadData.GetValueOrDefault<UInt32[]>());
                            break;
                        case "Int16":
                            var valueInt = ReadData.GetValueOrDefault<Int16>();
                            results.Add(valueInt.ToString());
                            break;
                        case "Boolean":
                            var valueBool = ReadData.GetValueOrDefault<Boolean>();
                            results.Add(valueBool.ToString());
                            break;
                        case "String":
                            var valueString = ReadData.GetValueOrDefault<String>();
                            results.Add(valueString.ToString());
                            break;
                        case "ByteString":
                            object valueByteString = ReadData.GetValueOrDefault<System.Byte[]>();
                            results.Add(valueByteString);
                            break;
                        default:
                            break;
                    }
                }
            }
            catch (Exception exception)
            {
                App.Log.Error("OPCConsumer::OpcArrayRead  - ERROR " + "1003" + "- " + exception.ToString());
            }
            return results;
        }
```

quinmars, 2020-04-25:
>You may benefit from a little known method named RegisterNodes.

Oh, that's very interesting! I need to try that out.

rebj1011, 2020-04-27:
Unfortunately, that doesn't help. It's just as fast as the function without the registered nodes.
Is there another way to speed it up?
Is our concept of reading so much data in one command perhaps not the right way?
Or should we redefine our nodes?
@quinmars Does it improve your communication?

awcullen, 2020-04-27:
If you want to try redefining the nodes, then I would try defining fewer nodes, with larger arrays.  Reading a fewer number of nodes might be a little bit faster.   I wouldn't expect a dramatic improvement.
Maybe you can ask this question from the Sinumerik group at the Siemens forum. 
https://support.industry.siemens.com/tf/us/en/

awcullen, 2020-05-07:
Please reopen at any time.

closed 2020-05-07 by awcullen

quinmars, 2020-05-10:
@rebj1011 I haven't tried it yet. But we do not have performance problems there. In our setup we read around 500-1000 (primitive) values and it takes around 200ms. The numbers, however, come from memory, I'd have to look up the exact values. We are using a 1500 soft plc.

rebj1011, 2020-05-11:
@quinmars How do you Read the 1000 "primitive" values?
@awcullen The Problem is i can't read lager Arrays, if i try to read more than 149 Values, i get no values.

rebj1011, 2020-05-11:
I can't Read more than 149 Byte in one read-Command. That's my problem. If i try to read more, i get an BadNodeIdUnknown (0x80340000) Error.