convertersystems/opc-ua-client #147: Read/Write Tag in AB plc

CSGstella, 2020-05-04:
I'm trying to communicate with an AB plc with his FactoryTalkLinxGateway. Is this library compatible with AB plc?
I'm normally not able to read any tag. but if i first read one tag with a test program from Siemens, and afterward I read again the same tag with my program it reads correctly.

were what I'm using to Read:
`public static async Task<string> Read(string variable)
        {
            //variable = "ns=3;s=variablename"
            string _res = "";
            var clientDescription = new ApplicationDescription
            {
                ApplicationName = "Workstation.UaClient.FeatureTests",
                ApplicationUri = $"urn:{System.Net.Dns.GetHostName()}:Workstation.UaClient.FeatureTests",
                ApplicationType = ApplicationType.Client
            };
            
            var channel = new UaTcpSessionChannel(
                clientDescription,
                null, // no x509 certificates
                new AnonymousIdentity(), // no user identity
                "opc.tcp://GSTELLA2017:4990/PLC_SAM_COM", // the public endpoint of a server at opcua.rocks.
                SecurityPolicyUris.None); // no encryption
            try
            {
                await channel.OpenAsync();
                var readRequest = new ReadRequest
                {
                    NodesToRead = new[] {
                        new ReadValueId {
                            NodeId = NodeId.Parse(variable),
                            AttributeId = AttributeIds.Value,
                        }
                    }
                };
                var readResult = await channel.ReadAsync(readRequest);

                if (readResult.Results[0].Value != null )
                {
                    var serverStatus = readResult.Results[0].Value.ToString();
                    _res= "VALUE: " + serverStatus;
                }
                else
                {
                    _res= "NO VALUE";
                }
                
                await channel.CloseAsync();
            }
            catch (Exception ex)
            {
                await channel.AbortAsync();
                Console.WriteLine(ex.Message);
            }
            return _res;
        }`
thanks in advance.

awcullen, 2020-05-04:
To find out the reason the result is null, you can check the StatusCode and Timestamp.
```
var readResult = await channel.ReadAsync(readRequest);
// Print the value, status code, and timestamp
Debug.WriteLine(readResult.Results[0]); 
```
Let me know the status code and I can look up the reason.

CSGstella, 2020-05-05:
here the result:

[image]

to give you better feedback i've crate a read request with multiple variables like this:
`var readRequest = new ReadRequest
                {
                    NodesToRead = new[] {
                            new ReadValueId {
                                NodeId = NodeId.Parse("ns=3;s=[SAM]PcRd.SAM[1].stsLoadingStatus"),
                                AttributeId = AttributeIds.Value
                            },
                            new ReadValueId {
                                NodeId = NodeId.Parse("ns=3;s=[SAM]PcRd.SAM[2].stsLoadingStatus"),
                                AttributeId = AttributeIds.Value
                            },
                            new ReadValueId {
                                NodeId = NodeId.Parse("ns=3;s=[SAM]PcRd.SAM[3].stsLoadingStatus"),
                                AttributeId = AttributeIds.Value
                            },
                            new ReadValueId {
                                NodeId = NodeId.Parse("ns=3;s=[SAM]PcRd.SAM[4].stsLoadingStatus"),
                                AttributeId = AttributeIds.Value
                            },
                            new ReadValueId {
                                NodeId = NodeId.Parse("ns=3;s=[SAM]PcRd.SAM[5].stsLoadingStatus"),
                                AttributeId = AttributeIds.Value
                            }
                    }
                };`

before to have this result i've read the first 3 variables with the siemens test program, it looks like it have "activate" the tag.

CSGstella, 2020-05-05:
here the result:
[image]

to give you better feedback i've crate a read request with multiple variables like this:

var readRequest = new ReadRequest { 
NodesToRead = new[] { 
new ReadValueId { NodeId = NodeId.Parse("ns=3;s=[SAM]PcRd.SAM[1].stsLoadingStatus"), AttributeId = AttributeIds.Value },
 new ReadValueId { NodeId = NodeId.Parse("ns=3;s=[SAM]PcRd.SAM[2].stsLoadingStatus"), AttributeId = AttributeIds.Value }, 
new ReadValueId { NodeId = NodeId.Parse("ns=3;s=[SAM]PcRd.SAM[3].stsLoadingStatus"), AttributeId = AttributeIds.Value }, 
new ReadValueId { NodeId = NodeId.Parse("ns=3;s=[SAM]PcRd.SAM[4].stsLoadingStatus"), AttributeId = AttributeIds.Value }, 
new ReadValueId { NodeId = NodeId.Parse("ns=3;s=[SAM]PcRd.SAM[5].stsLoadingStatus"), AttributeId = AttributeIds.Value } } };

before to have this result i've read the first 3 variables with the siemens test program, it looks like it have "activate" the tag.

awcullen, 2020-05-05:
StatusCode 0x80330000 is
// BadNodeIdInvalid - The syntax of the node id is not valid.

Very strange.

CSGstella, 2020-05-05:
I tried to read all 5 variables with the other test program, after that i can read all 5 values with my program as well. could it be related to the server somehow? maybe some parameter to be specify different form the standard one in the connection? i'm trying with the same code (just changing the connection parameter) to read/write on a S71500 and it works with no problem.
is there a document where i can find the error code meaning?

thanks!

awcullen, 2020-05-05:
I have an idea to try.  ReadRequest has a field MaxAge that is not set (default is 0.0) that is described as: 
Maximum age of the value to be read in milliseconds. The age of the value is based on the difference between the ServerTimestamp and the time when the Server starts processing the request. 
If the Server cannot meet the requested maxAge, it returns its “best effort” value rather than rejecting the request. This may occur when the time it takes the Server to process and return the new data value after it has been accessed is greater than the specified maximum age.
If maxAge is set to 0, the Server shall attempt to read a new value from the data source.
If maxAge is set to the max Int32 value or greater, the Server shall attempt to get a cached value.
Negative values are invalid for maxAge.

I would try setting MaxAge: 5000.0,  to allow the server to return a recently cached value of last 5 seconds.

CSGstella, 2020-05-05:
tried and same result, I get 
`{; status: 0x80330000; ts: 01-01-0001 00:00:00}`

awcullen, 2020-05-05:
You can find info about the UA services ( the requests, responses and paramters) from here:
https://reference.opcfoundation.org/v104/Core/docs/Part4/

The list of codes can be downloaded from here:
http://www.opcfoundation.org/UA/schemas/1.04/StatusCode.csv

awcullen, 2020-05-05:
Another parameter of ReadRequest is TimestampsToReturn (the default is TimestampsToReturn.Source).  
Please try:
TimestampsToReturn = TimestampsToReturn.None,
-or- 
 TimestampsToReturn = TimestampsToReturn.Both,

awcullen, 2020-05-05:
Which test program from Siemens are you using?  I wonder if this test program is creating a Subscription?

CSGstella, 2020-05-05:
tried, i get the same result.
one doubt about the "NONE", in my library the "TimestampsToReturn" variable have the following values:
[image]

CSGstella, 2020-05-05:
i'm using the test program that you can find here:
[42014088_OPC_UAClient_CODE_V12.zip](https://github.com/convertersystems/opc-ua-client/files/4581312/42014088_OPC_UAClient_CODE_V12.zip)

awcullen, 2020-05-05:
(Sorry,  should be TimestampsToReturn.Neither)  
I am downloading the Siemens client now.
I use a program called Wireshark to log the packets going between a client/server.

CSGstella, 2020-05-05:
ok, I tried also TimestampsToReturn.Neither, but same result.
I'll try tomorrow to sniff the packets going between the server and the client with Wireshark. I'll let you know.

CSGstella, 2020-05-06:
unfortunately for the moment i have no chance to test the packets going from the testApp to the Opc Server. the program and the opc Server are in the same PC.

awcullen, 2020-05-06:
I checked with Wireshark and found the Siemens client uses MaxAge=0.0, TimestampsToReturn = TimestampsToReturn.Neither.

When you use the Siemens client do you just read the attribute or do you drag the node into the lower box that creates a subscription?

CSGstella, 2020-05-06:
yes, checking the code they use always the parameters  MaxAge=0.0, TimestampsToReturn = TimestampsToReturn.Neither. I've copy it in my applicazion and I'm always creating the read request with this 2 parameters.
[image]
anyway i get the same result.

to "active" the tag I just click on the tag and I open the attribute on the right panel like this:
[image]

after that I can see it on my application.

awcullen, 2020-05-06:
Mystery.  I attached a function that creates a subscription and logs the output.  It should run for 60s and then end.  

```
public async Task CreateDataSubscription2()
{
    // describe this client application.
    var clientDescription = new ApplicationDescription
    {
        ApplicationName = "Workstation.UaClient.FeatureTests",
        ApplicationUri = $"urn:{System.Net.Dns.GetHostName()}:Workstation.UaClient.FeatureTests",
        ApplicationType = ApplicationType.Client
    };

    // create a 'UaTcpSessionChannel', a client-side channel that opens a 'session' with the server.
    var channel = new UaTcpSessionChannel(
        clientDescription,
        null, // no x509 certificates
        new AnonymousIdentity(), // no user identity
        "opc.tcp://GSTELLA2017:4990/PLC_SAM_COM", // the endpoint of a server
        SecurityPolicyUris.None // no encryption
        );

    try
    {
        // try opening a session and reading a few nodes.
        await channel.OpenAsync();

        Console.WriteLine($"Opened session with endpoint '{channel.RemoteEndpoint.EndpointUrl}'.");
        Console.WriteLine($"SecurityPolicy: '{channel.RemoteEndpoint.SecurityPolicyUri}'.");
        Console.WriteLine($"SecurityMode: '{channel.RemoteEndpoint.SecurityMode}'.");
        Console.WriteLine($"UserIdentityToken: '{channel.UserIdentity}'.");

        // build a CreateSubscriptionRequest. See 'OPC UA Spec Part 4' paragraph 5.13.2
        var req = new CreateSubscriptionRequest
        {
            RequestedPublishingInterval = 1000.0, // intervals are in milliseconds
            RequestedMaxKeepAliveCount = 30,
            RequestedLifetimeCount = 30 * 3,
            PublishingEnabled = true,
        };
        var res = await channel.CreateSubscriptionAsync(req);

        // the result will return the server's subscription id. You will needs this to 
        // add monitored items.
        var id = res.SubscriptionId;
        Console.WriteLine($"Created subscription '{id}'.");

        // build a CreateMonitoredItemsRequest. See 'OPC UA Spec Part 4' paragraph 5.12.2
        var req2 = new CreateMonitoredItemsRequest
        {
            SubscriptionId = id,
            TimestampsToReturn = TimestampsToReturn.Both,
            ItemsToCreate = new MonitoredItemCreateRequest[]
            {
                new MonitoredItemCreateRequest
                {
                    ItemToMonitor= new ReadValueId{ AttributeId= AttributeIds.Value, NodeId= NodeId.Parse("ns=3;s=[SAM]PcRd.SAM[1].stsLoadingStatus")},
                    MonitoringMode= MonitoringMode.Reporting,
                    RequestedParameters= new MonitoringParameters{ ClientHandle= 1, QueueSize= 1, DiscardOldest= true, SamplingInterval= 1000.0},
                },
                new MonitoredItemCreateRequest
                {
                    ItemToMonitor= new ReadValueId{ AttributeId= AttributeIds.Value, NodeId= NodeId.Parse("ns=3;s=[SAM]PcRd.SAM[2].stsLoadingStatus")},
                    MonitoringMode= MonitoringMode.Reporting,
                    RequestedParameters= new MonitoringParameters{ ClientHandle= 2, QueueSize= 1, DiscardOldest= true, SamplingInterval= 1000.0},
                },
                new MonitoredItemCreateRequest
                {
                    ItemToMonitor= new ReadValueId{ AttributeId= AttributeIds.Value, NodeId= NodeId.Parse("ns=3;s=[SAM]PcRd.SAM[3].stsLoadingStatus")},
                    MonitoringMode= MonitoringMode.Reporting,
[... 29 lines cut ...]
                    var dcns = pr.NotificationMessage.NotificationData.OfType<DataChangeNotification>();
                    foreach (var dcn in dcns)
                    {
                        foreach (var min in dcn.MonitoredItems)
                        {
                            Console.WriteLine($"sub: {pr.SubscriptionId}; handle: {min.ClientHandle}; value: {min.Value}");
                        }
                    }
                },
                ex => Console.WriteLine("Exception in publish response handler: {0}", ex.GetBaseException().Message)
            );

        // publish for 60 seconds and then close.
        await Task.Delay(60000);

        Console.WriteLine($"\nClosing session '{channel.SessionId}'.");
        await channel.CloseAsync();
    }
    catch (Exception ex)
    {
        await channel.AbortAsync();
        Console.WriteLine(ex.Message);
    }
}
```

awcullen, 2020-05-06:
I was unable to setup a instance of FactoryTalk Linx Gateway with a simulation PLC for testing.
Would you mind running the following test? It duplicates the actions of the Siemens client when browsing:
```
public static async Task BrowseAndRead()
{
    var clientDescription = new ApplicationDescription
    {
        ApplicationName = "Workstation.UaClient.FeatureTests",
        ApplicationUri = $"urn:{System.Net.Dns.GetHostName()}:Workstation.UaClient.FeatureTests",
        ApplicationType = ApplicationType.Client
    };

    var channel = new UaTcpSessionChannel(
        clientDescription,
        null, // no x509 certificates
        new AnonymousIdentity(), // no user identity
        "opc.tcp://GSTELLA2017:4990/PLC_SAM_COM", // the public endpoint of a server at opcua.rocks.
        SecurityPolicyUris.None); // no encryption
    try
    {
        await channel.OpenAsync();

        // Browse the object SAM
        var browseRequest = new BrowseRequest
        {
            NodesToBrowse = new[] {
            new BrowseDescription { NodeId = NodeId.Parse("ns=3;s=[SAM]PcRd.SAM"), ReferenceTypeId = NodeId.Parse(ReferenceTypeIds.HierarchicalReferences), ResultMask = (uint)BrowseResultMask.All, NodeClassMask = (uint)NodeClass.Unspecified, BrowseDirection = BrowseDirection.Forward, IncludeSubtypes = true }
            },
            RequestedMaxReferencesPerNode = 100
        };
        var browseResponse = await channel.BrowseAsync(browseRequest).ConfigureAwait(false);
        foreach (var rd in browseResponse.Results[0].References)
        {
            Console.WriteLine("   + {0}, {1}, {2}, {3}", rd.DisplayName, rd.BrowseName, rd.NodeClass, rd.NodeId);
        }

        // Browse the object SAM[1]
        browseRequest = new BrowseRequest
        {
            NodesToBrowse = new[] {
            new BrowseDescription { NodeId = NodeId.Parse("ns=3;s=[SAM]PcRd.SAM[1]"), ReferenceTypeId = NodeId.Parse(ReferenceTypeIds.HierarchicalReferences), ResultMask = (uint)BrowseResultMask.All, NodeClassMask = (uint)NodeClass.Unspecified, BrowseDirection = BrowseDirection.Forward, IncludeSubtypes = true }
            },
            RequestedMaxReferencesPerNode = 100
        };
        browseResponse = await channel.BrowseAsync(browseRequest).ConfigureAwait(false);
        foreach (var rd in browseResponse.Results[0].References)
        {
            Console.WriteLine("   + {0}, {1}, {2}, {3}", rd.DisplayName, rd.BrowseName, rd.NodeClass, rd.NodeId);
        }

        // Read variable SAM[1].stsLoadingStatus
        var readRequest = new ReadRequest
        {
            NodesToRead = new[] {
            new ReadValueId {
                NodeId = NodeId.Parse("ns=3;s=[SAM]PcRd.SAM[1].stsLoadingStatus"),
                AttributeId = AttributeIds.Value,
                }
            },
            TimestampsToReturn = TimestampsToReturn.Neither
        };
        var readResult = await channel.ReadAsync(readRequest);

        Console.WriteLine(readResult.Results[0]);

        await channel.CloseAsync();
    }
    catch (Exception ex)
    {
        await channel.AbortAsync();
        Console.WriteLine(ex.Message);
    }
}
```

CSGstella, 2020-05-07:
I've test the task CreateDataSubscription2 and it works!
It create an opc group in the server so I van read the variable.

thanks a lot!

closed 2020-05-07 by CSGstella