convertersystems/opc-ua-client #148: ask for advice: associate MonitoredItem Name to MonitoredItemNotification.Value

FedericoBgn, 2020-05-04:
Hi,
sorry if this is not the right place to ask for help. New to GitHub.
I work on PLC side, but I am a C# beginner. I'm trying to learn the basics of OPC UA. 
I looked at opc-ua-samples console application and OpcUaFeatureTests.
I created a class that performs the task of creating a connection, reading and writing the value of a node,  creating a subscription and inserting in it some MonitordItems,. 
The subscription is working and I can retrieve values, on data change, from TIA15.1 WinCC Simulator.
But now I need to get the couple MonitoredItem Name + MonitoredItemNotification Value. Can I get both this data in a console application?  I’m a bit suck in retrieving the name of the MonitoredItem, source of the Notification.
token = channel.Where(pr => pr.SubscriptionId == id).Subscribe(pr =>
    {
    // loop thru all the data change notifications
    var dcns = pr.NotificationMessage.NotificationData.OfType<DataChangeNotification>();
    foreach (var dcn in dcns)
    {
        foreach (var min in dcn.MonitoredItems)
        {
            // here I can get MonitoredItemNotification.Value,
            // but also I need the name of the MonitoredItem whose valuee has changed.
            Console.Write($"value: {min.Value}; type : {min.GetType().ToString()}; sub: {pr.SubscriptionId};      handle: {min.ClientHandle}");
        }
    }
});
`
Thanks in Advance.

awcullen, 2020-05-04:
Good question!  

The  MonitoredItemNotification contains the field "ClientHandle" which is an integer (UInt32). You assign the ClientHandle when you call CreateMonitoredItems.  When I create my subscription and monitored items I assign a unique value to the ClientHandle.  When I handle the Notification, I can use the ClientHandle to look up the related data from a dictionary.

HTH

FedericoBgn, 2020-05-05:
Thanks very much.
I will follow your advice. I gave a quick look at my application and I'm sure I can use ClientHandle as suggested.
Great framework.

closed 2020-05-05 by FedericoBgn