convertersystems/opc-ua-client #151: create channel with using username and password or certificate

shabshekan, 2020-05-28:
I try to create channel with username & password. I try in this way,

 var channel = new UaTcpSessionChannel(
                    clientDescription,
                    null, // no x509 certificates
                   new UserNameIdentity("user1", "pass123") ,
                    "opc.tcp://127.0.0.1:4861",
                    SecurityPolicyUris.None);
Is it correct or I need to do more?
Could you please provide sample for using certificate to create channel?

awcullen, 2020-05-28:
Your code is correct.
To add a client certificate you can use:
```
var clientDescription = new ApplicationDescription
{
    ApplicationName = "Workstation.UaClient.FeatureTests",
    ApplicationUri = $"urn:{System.Net.Dns.GetHostName()}:Workstation.UaClient.FeatureTests",
    ApplicationType = ApplicationType.Client
};

// place to store certificates
var certificateStore = new DirectoryStore("./pki");

// create a channel that opens a 'session' with the server.
var channel = new UaTcpSessionChannel(
   clientDescription,
   certificateStore,
   new UserNameIdentity("user1", "pass123") ,
   "opc.tcp://127.0.0.1:4861"); 
```
Initially, the "./pki" directory will be empty.  The channel will create a new certificate and private key and store it in the directory for next time.  Your server may reject the client certificate initially.  You will have to configure the server to accept the client certificate.
  
You can find more samples like this at https://github.com/convertersystems/OpcUaFeatureTests

shabshekan, 2020-05-29:
Thank you so much @awcullen

closed 2020-06-04 by awcullen