convertersystems/opc-ua-client #160: Error when comparing server certificate for connection and session

wazzamatazz, 2020-08-17:
When connecting to a particular OPC UA server, I get an error in `UaTcpSessionChannel.OnOpenAsync`:

```
System.ArgumentNullException: Value cannot be null.
Parameter name: second
   at System.Linq.Enumerable.SequenceEqual[TSource](IEnumerable`1 first, IEnumerable`1 second, IEqualityComparer`1 comparer)
   at Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel.<OnOpenAsync>d__48.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Workstation.ServiceModel.Ua.Channels.CommunicationObject.<OpenAsync>d__33.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at Workstation.ServiceModel.Ua.Channels.CommunicationObject.<OpenAsync>d__33.MoveNext()
```

Looking at the source, I can see that `SequenceEqual` is used to [ensure that the connection and session certificates are identical](https://github.com/convertersystems/opc-ua-client/blob/e68429a995717321d7362d46e5c83a73d0c33190/UaClient/ServiceModel/Ua/Channels/UaTcpSessionChannel.cs#L341). The error is therefore occurring because the OPC UA server is returning `null` for the `ServerCertificate` property on the `CreateSessionResponse` object.

The OPC UA connection is configured to use `None` for the server security policy, and also to use an anonymous user identity. From what I understand from the OPC UA spec (admittedly limited!), it seems legitimate that the server might not include the certificate in the create session response, because clients are supposed to ignore the server certificate if the security policy is `None` and if none of the identity policies on the endpoint require encryption: 

_OPC Unified Architecture - Part 4: Services, section 5.6.2.2 (`CreateSession` respoonse properties)_

_`serverCertificate`: If the securityPolicyUri is NONE and none of the UserTokenPolicies requires encryption, the Client shall ignore the ApplicationInstanceCertificate._

Would an appropriate fix be to add an additional check so that the certificate check is not performed if the channel's endpoint only lists user token policies that specify `None` as the security policy? Or would it be sufficient to skip the certificate check when an anonymous user identity is being used? Happy to submit a PR if I can get some clarification on which approach would be preferred.

awcullen, 2020-08-17:
Hi wazzamatazz, 

The PR looks good.  Can you share which brand of OPC server you are connecting to?  Also, when connecting, the client gets the ServerCertificate twice really. Once when it calls GetEndpoints, and a second time when calling CreateSession.  Do you know if your OPC server is returning null for ServerCertificate in both responses?  

BR, 
Andrew

wazzamatazz, 2020-08-18:
Hi,

It's a [Prediktor APIS Historian](https://www.prediktor.no/industrial-info-management/apis-store-historian/) that I'm connecting to. Based on the nature of the error (and reviewing the source for `UaTcpSessionChannel`), I think that the certificate is being returned with the endpoint information but is not being returned when creating the session, because the certificate comparison is currently only performed if the certificate returned in the endpoint information is non-null.

I'm trying to clarify with them what the user token security policies are set to when the server security policy is `None`, to verify if my PR would actually fix the issue (in case they are still specifying a user token security policy that requires encryption).

As an aside, I also wanted to say that I've found this library really easy to use, so many thanks for the effort that's gone into it!

Thanks,
Graham

awcullen, 2020-08-20:
Thanks Graham, PR looks good.

wazzamatazz, 2020-08-21:
Great. I think the vendor is actually specifying a user token security policy of `None` in this instance, so the PR should fix the connection issue.

closed 2020-08-21 by wazzamatazz