convertersystems/opc-ua-client #164: Error receiving response. Unable to read beyond the end of the stream

dojo90, 2020-09-09:
I have a Siemens 15xx PLC with the following `struct`:

[image]

[image]

I made the following `Structure` class:

```csharp
[BinaryEncodingId("nsu=http://www.siemens.com/simatic-s7-opcua;s=TE_\"HTOPC_EquipmentModule_V1\"")]
internal class EquipmentModuleStructure : Structure
{
	public override void Decode(IDecoder decoder)
	{

	}

	public override void Encode(IEncoder encoder)
	{
		
	}
}
```

Everytime when I try to read the `node` I get the following error and my `channel` breaks:

[image]

> Error receiving response. Unable to read beyond the end of the stream

Is there something wrong with my setup? The funny thing is that I can read out the status, for example. The error also occurs when I have no logic within the Decode() (see example above).

[image]

quinmars, 2020-09-09:
The `Decode` and `Encode` methods are not optional. You have to implement them correctly. That means you have to read/write the value with the correct type in the correct order. Else the internal stream of the encoder/decoder will not be advanced and the encoder/decoder will resume the parsing/writing at the wrong position in the stream.

dojo90, 2020-09-09:
This was the solution 🙂👍

closed 2020-09-09 by dojo90