convertersystems/opc-ua-client #165: Monitoring System.Double[]

dmkuntz, 2020-09-14:
I am attempting to monitor a System.Double[] datatype from an OPC server.  I understand that these type of arrays are SafeArrays and not full .NET arrays.

My monitor works successfully, and is set up like this:

`       var dcns = pr.NotificationMessage.NotificationData.OfType<DataChangeNotification>();
            foreach (var dcn in dcns)
            {
                foreach (var min in dcn.MonitoredItems)
                {
                        Console.WriteLine(min.Value);
                }
            }

and I retrieve on the console log
`System.Double[]; status: 0x00000000; ts: 9/14/2020 7:42:55 PM`

I have attempted to access the values behind System.Double[] in accordance to this Stack Overflow article:

https://stackoverflow.com/questions/23125903/what-does-system-double-mean#:~:text=A%20System.,heavily%20optimized%20in%20the%20CLR.

However, because there is no length or rank of this object, I cannot convert it into a usable array or an array type.  I also cannot iterate through it.  What can I do to access the array, or are not all necessary attributes being imported?  Thank you for your help

quinmars, 2020-09-14:
`min.Value` is of the type `DataValue`. To access the actual data, you can use the `Value` property of the `DataValue` class. This is of the type `object` so you have to cast it to the expected type. For example:

```csharp
foreach (var dcn in dcns)
{
    foreach (var min in dcn.MonitoredItems)
    {
        if (min.Value.Value is double[] array)
        {
            // ... use array here
        }
        else
        {
            Console.WriteLine("Oops, not a double array.");
        }
    }
}
```

dmkuntz, 2020-09-14:
Thanks for the quick response!  I did the following and it worked.  I needed the min.Value.Value instead of min.Value as you said

```
if (min.Value.Value is double[] array)
{
    var doubleArray = min.Value.Value as double[];
    Console.WriteLine(doubleArray.Length);
     ...
}
```

closed 2020-09-14 by dmkuntz

quinmars, 2020-09-14:
You are welcome. Just a sidenote: you can use the variable `array` directly.

```csharp
if (min.Value.Value is double[] array)
{
    Console.WriteLine(array.Length);
     ...
}
```