convertersystems/opc-ua-client #166: "Timeout occurred while processing the request." on Method Call request

kaasmdfs, 2020-09-16:
While trying to call a method from an OPC server with a custom type I always receive a "Timeout occurred while processing the request." error.  Below is the code for the request, it follows the logic you used for the method call in the IntegrationTests class.

```
  protected override async Task ExecuteAsync(CancellationToken stoppingToken)
   { 

            // describe this client application.
            var clientDescription = new ApplicationDescription
            {
                ApplicationName = "Workstation.UaClient.FeatureTests",
                ApplicationUri = $"urn:{System.Net.Dns.GetHostName()}:Workstation.UaClient.FeatureTests",
                ApplicationType = ApplicationType.Client
            };

            var channel = new UaTcpSessionChannel(
              this.localDescription,
              null,
              new AnonymousIdentity(),
             "opc.tcp://192.168.0.254:4840",
              SecurityPolicyUris.None,
              loggerFactory: this.loggerFactory,
              additionalTypes: new[] { typeof(ScanSettings) });
            try
            {
                // try opening a session and reading a few nodes.
                await channel.OpenAsync();

                Console.WriteLine($"Opened session with endpoint '{channel.RemoteEndpoint.EndpointUrl}'.");
                Console.WriteLine($"SecurityPolicy: '{channel.RemoteEndpoint.SecurityPolicyUri}'.");
                Console.WriteLine($"SecurityMode: '{channel.RemoteEndpoint.SecurityMode}'.");
                Console.WriteLine($"UserIdentityToken: '{channel.UserIdentity}'.");
                var readRequest = new ReadRequest { NodesToRead = new[] { new ReadValueId { NodeId = NodeId.Parse("ns=4;i=6030"), AttributeId = AttributeIds.Value } } };
                var readResult = await channel.ReadAsync(readRequest);
                var serverStatus = readResult.Results[0].GetValueOrDefault<ServerStatusDataType>();

                var set = new ScanSettings
                {
                    Duration = 500,
                    Cycles = 10
                };
                var request = new CallRequest
                {
                    MethodsToCall = new[] {
                        new CallMethodRequest
                        {
                            ObjectId =  NodeId.Parse("ns=2;i=5001"),
                            MethodId =  NodeId.Parse("ns=4;i=7010"),
                            InputArguments = new [] { new ExtensionObject(set)}.ToVariantArray()
                        }
                    }
                };
 
                var response = await channel.CallAsync(request);
                var result = response.Results[0].OutputArguments[0].GetValueOrDefault<ScanSettings>();

                Console.WriteLine($"\nClosing session '{channel.SessionId}'.");
                await channel.CloseAsync();
            }
            catch (Exception ex)
            {
                await channel.AbortAsync();
                Console.WriteLine(ex.Message);
            }
        }

```
```

    [DataTypeId("ns=3;i=6044")]
    [BinaryEncodingId("ns=3;i=5015")]
    internal class ScanSettings : Structure
    {
        public uint Duration { get; set; }
        public int Cycles { get; set; }
        public override void Encode(IEncoder encoder)
        {
            encoder.WriteUInt32("Duration", this.Duration);
            encoder.WriteInt32("Cycles", this.Cycles);
        }

        public override void Decode(IDecoder decoder)
        {
            this.Duration = decoder.ReadUInt32("Duration");
            this.Cycles = decoder.ReadInt32("Cycles");
        }

        public override string ToString() => $"{{ Duration={this.Duration}; Cycles={this.Cycles}; }}";
    }
```

Any help would be awesome.
Thank you

awcullen, 2020-09-17:
Code looks good. Could you share the log?
I see that you are calling a method "ns=4;i=7010" of parent object "ns=2;i=5001".
This method takes a single argument of ScanSettings ("ns=3;i=6044") and returns a single argument of type ScanSettings.

Could you share a shot of UAExpert or your favorite tool making a call?  Similar to:
[image]

kaasmdfs, 2020-09-17:
Here is a screenshot of the ScanSettings. 
[image]

And a screen of the result.
[image]

And one last image of the hierarchy in UaExpert.  'DeviceSet' is  'ns=2;i=5001'  and 'Scan' is  "ns=4;i=7010" but I have tried the id for Read_point_1 as well.

Finally here is the output from the console.
[image]

Thanks for the incite.

awcullen, 2020-09-19:
To avoid the decoding error, I believe we will need to define the StatusSettings structure completely.   Let me show you what I do:
I use UAExpert to discover the fields of the structure.  UA Servers document the fields of structures in two ways.  The original way is to store a TypeDictionary file in a location that the client can read.  I use UAExpert to locate the folder "OPC Binary".  

[image]

Under "OPC Binary", I found variables "BA", "Demo", and "Demo2". Each of these variables stores a value of type ByteString.  

[image]

I ask that you copy the bytes from the Value of the first variable and paste it into a webpage at https://hexdecoder.com/

[image]

This uncovers that the ByteString is an XML file that defines the fields of  custom structures. 

Please share what you find.

kaasmdfs, 2020-09-21:
So I followed your suggestion and ran into the same error message.  The AutoID object is the only one that had a Value in it's attributes so I opened that.
[image]

This is the XML result of that binary string.
[image]

So I updated the code to be.

[image]

One thing I'm not sure how to make sense of is the LocationTypeSpecified property, because it is a bit and has an Enum attached to its value.

I ran a wireshark scan of the result coming from UA Expert and from my code and the only difference is the ByteString being sent.

Thank you for the help.

quinmars, 2020-09-21:
According to https://reference.opcfoundation.org/v104/AutoID/v101/docs/9.3.7/, the `LocationTypeSpecified` needs to be an enumeration instead of a boolean and the `Duration` needs to be a `double`. But I have no clue what the `Bit` and `SwitchField` does mean. Hopefully Andrew has more insight into that.

kaasmdfs, 2020-09-21:
Got a solution.  By adding in a enum with the values that were needed the call succeeded.

```
    [DataTypeId("ns=3;i=6044")]
    [BinaryEncodingId("ns=3;i=5015")]
    internal class ScanSettings : Structure
    {
        public double Duration { get; set; }
        public int Cycles { get; set; }
        public bool DataAvailable { get; set; }
        public LocationTypeEnumeration LocationTypeSpecified { get; set; }
        public override void Encode(IEncoder encoder)
        {
            encoder.WriteDouble("Duration", this.Duration);
            encoder.WriteInt32("Cycles", this.Cycles);
            encoder.WriteBoolean("DataAvailable", this.DataAvailable);
            encoder.WriteEnumeration("LocationTypeSpecified", this.LocationTypeSpecified);
        }

        public override void Decode(IDecoder decoder)
        {
            this.Duration = decoder.ReadDouble("Duration");
            this.Cycles = decoder.ReadInt32("Cycles");
            this.DataAvailable = decoder.ReadBoolean("DataAvailable");
        }

        public override string ToString() => $"{{ Duration={this.Duration}; Cycles={this.Cycles}; DataAvailable={this.DataAvailable}; LocationTypeSpecified={this.LocationTypeSpecified};}}";
    }
    public enum LocationTypeEnumeration
    {
        NMEA = 0,
        LOCAL = 1,
        WGS84 = 2,
        NAME = 3
    }
```

Thank you all for help and leading me in the right direction.

awcullen, 2020-09-21:
Try this out.   I got the idea from https://reference.opcfoundation.org/v104/Core/docs/Part6/5.2.7/
```

    [DataTypeId("nsu=http://opcfoundation.org/UA/AutoID/;i=3010")]
    [BinaryEncodingId("nsu=http://opcfoundation.org/UA/AutoID/;i=5015")]
    internal class ScanSettings : Structure
    {
        public Double Duration { get; set; }
        public Int32 Cycles { get; set; }
        public Boolean DataAvailable { get; set; }
        public Int32? LocationType { get; set; }

        public override void Encode(IEncoder encoder)
        {
            UInt32 optionFlags = 0;
            if (this.LocationType.HasValue)
            {
                optionFlags |= 1;
            }
            encoder.WriteUInt32("", optionFlags);
            encoder.WriteDouble("Duration", this.Duration);
            encoder.WriteInt32("Cycles", this.Cycles);
            encoder.WriteBoolean("DataAvailable", this.DataAvailable);
            if ((optionFlags & 1) != 0)
            {
                encoder.WriteInt32("LocationType", (int)this.LocationType);
            }
        }

        public override void Decode(IDecoder decoder)
        {
            UInt32 optionFlags = decoder.ReadUInt32("");
            this.Duration = decoder.ReadDouble("Duration");
            this.Cycles = decoder.ReadInt32("Cycles");
            this.DataAvailable = decoder.ReadBoolean("DataAvailable");
            if ((optionFlags & 1) != 0)
            {
                this.LocationType = decoder.ReadInt32("LocationType");
            } 
            else
            {
                this.LocationType = null;
            }
        }

        public override string ToString() => $"{{ Duration={this.Duration}; Cycles={this.Cycles}; }}";
    }
```

kaasmdfs, 2020-09-23:
I can get the call to succeed but now the reply from the opc server can't seem to parse the return result.  Below is the xml documentation for the return.
```

    <!--Call Input-->
    <opc:StructuredType BaseType="ua:ExtensionObject" Name="ScanSettings">
        <opc:Field TypeName="opc:Bit" Name="LocationTypeSpecified"/>
        <opc:Field Length="31" TypeName="opc:Bit" Name="Reserved1"/>
        <opc:Field TypeName="opc:Double" Name="Duration"/>
        <opc:Field TypeName="opc:Int32" Name="Cycles"/>
        <opc:Field TypeName="opc:Boolean" Name="DataAvailable"/>
        <opc:Field SwitchField="LocationTypeSpecified" TypeName="tns:LocationTypeEnumeration" Name="LocationType"/>
    </opc:StructuredType>
    <!-- End Call Input-->
     <!-- Call Result and sub classes-->
      <opc:StructuredType BaseType="tns:ScanResult" Name="RfidScanResult">
        <opc:Field TypeName="opc:Bit" Name="LocationSpecified"/>
        <opc:Field Length="31" TypeName="opc:Bit" Name="Reserved1"/>
        <opc:Field SourceType="tns:ScanResult" TypeName="opc:CharArray" Name="CodeType"/>
        <opc:Field SourceType="tns:ScanResult" TypeName="tns:ScanData" Name="ScanData"/>
        <opc:Field SourceType="tns:ScanResult" TypeName="opc:DateTime" Name="Timestamp"/>
        <opc:Field SwitchField="LocationSpecified" SourceType="tns:ScanResult" TypeName="tns:Location" Name="Location"/>
        <opc:Field TypeName="opc:Int32" Name="NoOfSighting"/>
        <opc:Field LengthField="NoOfSighting" TypeName="tns:RfidSighting" Name="Sighting"/>
    </opc:StructuredType>
     <opc:StructuredType BaseType="ua:ExtensionObject" Name="RfidSighting">
        <opc:Field TypeName="opc:Int32" Name="Antenna"/>
        <opc:Field TypeName="opc:Int32" Name="Strength"/>
        <opc:Field TypeName="opc:DateTime" Name="Timestamp"/>
        <opc:Field TypeName="opc:Int32" Name="CurrentPowerLevel"/>
    </opc:StructuredType>
  <opc:StructuredType BaseType="ua:ExtensionObject" Name="ScanResult">
        <opc:Field TypeName="opc:Bit" Name="LocationSpecified"/>
        <opc:Field Length="31" TypeName="opc:Bit" Name="Reserved1"/>
        <opc:Field TypeName="opc:CharArray" Name="CodeType"/>
        <opc:Field TypeName="tns:ScanData" Name="ScanData"/>
        <opc:Field TypeName="opc:DateTime" Name="Timestamp"/>
        <opc:Field SwitchField="LocationSpecified" TypeName="tns:Location" Name="Location"/>
    </opc:StructuredType>
    <opc:StructuredType BaseType="ua:Union" Name="Location">
        <opc:Field TypeName="opc:UInt32" Name="SwitchField"/>
        <opc:Field SwitchField="SwitchField" TypeName="opc:CharArray" SwitchValue="1" Name="NMEA"/>
        <opc:Field SwitchField="SwitchField" TypeName="tns:LocalCoordinate" SwitchValue="2" Name="Local"/>
        <opc:Field SwitchField="SwitchField" TypeName="tns:WGS84Coordinate" SwitchValue="3" Name="WGS84"/>
        <opc:Field SwitchField="SwitchField" TypeName="opc:CharArray" SwitchValue="4" Name="Name"/>
    </opc:StructuredType>

    <opc:StructuredType BaseType="ua:Union" Name="ScanData">
        <opc:Field TypeName="opc:UInt32" Name="SwitchField"/>
        <opc:Field SwitchField="SwitchField" TypeName="opc:ByteString" SwitchValue="1" Name="ByteString"/>
        <opc:Field SwitchField="SwitchField" TypeName="opc:CharArray" SwitchValue="2" Name="String"/>
        <opc:Field SwitchField="SwitchField" TypeName="tns:ScanDataEpc" SwitchValue="3" Name="Epc"/>
        <opc:Field SwitchField="SwitchField" TypeName="ua:Variant" SwitchValue="4" Name="Custom"/>
    </opc:StructuredType>
    <opc:StructuredType BaseType="ua:ExtensionObject" Name="LocalCoordinate">
        <opc:Field TypeName="opc:Double" Name="X"/>
        <opc:Field TypeName="opc:Double" Name="Y"/>
        <opc:Field TypeName="opc:Double" Name="Z"/>
        <opc:Field TypeName="opc:DateTime" Name="Timestamp"/>
        <opc:Field TypeName="opc:Double" Name="DilutionOfPrecision"/>
        <opc:Field TypeName="opc:Int32" Name="UsefulPrecicision"/>
    </opc:StructuredType>
     <opc:StructuredType BaseType="ua:ExtensionObject" Name="WGS84Coordinate">
        <opc:Field TypeName="opc:CharArray" Name="N_S_Hemisphere"/>
        <opc:Field TypeName="opc:Double" Name="Latitude"/>
        <opc:Field TypeName="opc:CharArray" Name="E_W_Hemisphere"/>
        <opc:Field TypeName="opc:Double" Name="Longitude"/>
        <opc:Field TypeName="opc:Double" Name="Altitude"/>
        <opc:Field TypeName="opc:DateTime" Name="Timestamp"/>
        <opc:Field TypeName="opc:Double" Name="DilutionOfPrecision"/>
        <opc:Field TypeName="opc:Int32" Name="UsefulPrecisionLatLon"/>
[... 247 lines cut ...]
    internal class ScanDataEpc : Structure
    {
        public UInt16 pC { get; set; }
        public byte[] uId { get; set; }
        public UInt16 xPC_W1 { get; set; }
        public UInt16 xPC_W2 { get; set; }
        public override void Encode(IEncoder encoder)
        {
            encoder.WriteUInt16("pC", this.pC);
            encoder.WriteUInt16("xPC_W1", this.xPC_W1);
            encoder.WriteUInt16("xPC_W2", this.xPC_W2);
            encoder.WriteByteString("uId", this.uId);
        }

        public override void Decode(IDecoder decoder)
        {
            this.pC = decoder.ReadUInt16("pC");
            this.xPC_W1 = decoder.ReadUInt16("xPC_W1");
            this.xPC_W2 = decoder.ReadUInt16("xPC_W2");
            this.uId = decoder.ReadByteString("uId");
        }

        public override string ToString() => $"{{ pC={this.pC}; xPC_W1={this.xPC_W1}; xPC_W2={this.xPC_W2}; uId={this.uId};}}";
    }
```

quinmars, 2020-10-02:
@kaasmdfs, you must not leave implementations out and you should use the encoding mask as shown by Andrew. Unions are yet another beast, that need a special treatment. I'm working at the moment on a generator, that generates the classes and enums from the Nodeset2 file. It is not finished yet, but I'll let you know when.

kaasmdfs, 2020-10-02:
Thanks everyone for the help, I managed to get a solution that used a version of what Andrew showed for the Unions that solved that one issue.

```
   [DataTypeId("ns=3;i=6034")]
    [BinaryEncodingId("ns=3;i=5013")]
    public class Location : Union
    {
        public int? SwitchField { get; set; }
        public string NMEA { get; set; }
        public LocalCoordinate Local { get; set; }
        public WGS84Coordinate WGS84 { get; set; }
        public string Name { get; set; }

        public override void Encode(IEncoder encoder)
        {
            encoder.WriteInt32("SwitchField", this.SwitchField.Value);
        }

        public override void Decode(IDecoder decoder)
        {
            this.SwitchField = decoder.ReadInt32("SwitchField");
            switch (SwitchField)
            {
                case 0:

                    break;
                case 1:
                    this.NMEA = decoder.ReadString("NMEA");
                    break;
                case 2:
                    this.Local = decoder.ReadExtensionObject<LocalCoordinate>("Local");
                    break;
                case 3:
                    this.WGS84 = decoder.ReadExtensionObject<WGS84Coordinate>("WGS84");
                    break;
                case 4:
                    this.Name = decoder.ReadString("Name");
                    break;
                default:
                    this.SwitchField = 0;
                    break;
            }

        }

     
    }
```

The rest of it was solved by adding a blank ReadInt32 at the beginning of the decoding, not sure why it was needed exactly, and I also had a few properties out of order from their specification.  I did not realize that the property order had to be exactly as defined in the XML.

quinmars, 2020-10-03:
Nice to hear that it works now for you. My generator supports now structures with optional fields and unions. I couldn't, however, test it, since I do not have access to an AutoID device. You find the file for it here:

https://github.com/quinmars/UaTypeGenerator/blob/master/Ua.AutoID/Opc.Ua.AutoID.NodeSet2.cs

and the DI types here (I don't know if you really need them):

https://github.com/quinmars/UaTypeGenerator/blob/master/Ua.DI/Opc.Ua.Di.NodeSet2.cs

I welcome any feedback, whether positive or negative.

closed 2021-02-07 by awcullen