convertersystems/opc-ua-client #169: Data type libraries for companion specifications

quinmars, 2020-10-03:
As already mention in another issue, I have written a [generator](https://github.com/quinmars/UaTypeGenerator) that generates C# classes from Nodest2.xml files. And now I'm quite satisfied with its output. My first idea was to utilize the new roslyn feature called "source generator". That would have some benefits, because you can access the type structure of the target project, and for example inspect its dependency assemblies or not generating classes that are already written by hand. But it turns out that the "output" of such projects is intentionally not written to disk. That's why I decided to go the traditional way, that is a plain console application.

With this generator we could provide nuget packages for all companion specifications found at https://github.com/OPCFoundation/UA-Nodeset, i.e., having a `opc-ua-nodesets` or `opc-ua-companions` What do you think about that? If you like we can also transfer the generator to convertersystems.

P.S. While writing the generator I noticed some minor issues that could be improved by UaClient. I will open a new issues for them soon.

awcullen, 2020-10-10:
Your generator is well written, and outputs clear source code.  It will be fantastic to have the many companion specs "pre-generated" as NuGet packages, ready to be included into a project.   I will be happy to incorporate any changes that makes this easier.  

Do you think I should merge the "TypeLibraryAttribute" and associated code that finds the attribute and registers the types with the encoder/decoder?

quinmars, 2020-10-12:
I'm glad to hear you like it. 

> Do you think I should merge the "TypeLibraryAttribute" and associated code that finds the attribute and registers the types with the encoder/decoder?

Yes, please. Although I do not need it for the code generation, I will use `TypeLibrary` attribute in code so that the user can easly use all exposed types.