convertersystems/opc-ua-client #172: Custom Structure in View Model

invinctus, 2020-10-15:
I am talking to a bosch OPC server and I am trying to map a property in my view model to a structure.

    [Subscription(endpointUrl: "opc", publishingInterval: 2000, keepAliveCount: 20)]
    public class CNCSystemVariables : SubscriptionBase
    {
        [MonitoredItem("ns=27;s=NC.SystemData,/MachineData/ServiceLocation")]
        public Cartesian ServiceLocation{
            get => serviceLocation;
            set => SetProperty(ref serviceLocation, value);
        }
    }

    [DataTypeId("ns=28;s=Cartesian_ty")]
    public class Cartesian : Structure
    {
        public float X { get; set; }
        public float Y { get; set; }
        public float Z { get; set; }

        public void Decode(IDecoder decoder) {
            this.X = decoder.ReadFloat("X");
            this.Y = decoder.ReadFloat("Y");
            this.Z = decoder.ReadFloat("Z");
        }

        public void Encode(IEncoder encoder) {
            encoder.WriteFloat("X", X);
            encoder.WriteFloat("Y", Y);
            encoder.WriteFloat("Z", Z);
        }
    }

This gives me a cannot convert ExtensionObject to Cartesian error.

I have several of these locations and I am trying to avoid subscribing to each individual property.

What am I missing?
Is this possible?

quinmars, 2020-10-15:
You are missing the `BinaryEncodingId` on the cartesian data type, note, that this is usually different to the data type id. If you are not already doing it, you have to pass the `type` in the additional type list to the channel.

invinctus, 2020-10-15:
how do i find out what the binary encoding id for this data type is?

quinmars, 2020-10-17:
If you have the Nodeset2.xml file you can find it there. Here is an example for an data type with the ID `ns=1;i=3019`, which has a binary encoding ID of `ns=1;i=5029`:

[image]

If you have the nodeset xml file you can also try to use the [UaTypeGenerator](https://github.com/quinmars/UaTypeGenerator).

awcullen, 2020-10-18:
Using a tool like UaExpert, you can browse the DataTypes folder for your structure. When you find it, there should be a "HasEncoding" reference named "Default Binary".  Hover over "Default Binary" to discover the NodeId.

Then add the attribute to the class declaration, for example:

[BinaryEncodingId("nsu=http://www.unifiedautomation.com/DemoServer/;i=5054")]    

[image]

invinctus, 2020-10-19:
I cannot find the type in the DataTypes section of UAExpert and the types that are present don't have a HasEncoding property.
[image]
If I subscribe to the property in UaExpert it shows up as an ExtensionObject

awcullen, 2020-10-20:
I believe you may find the type by looking under Root...Types...DataTypes...BaseDateType...Structure....Cartesian

But if the type is not found, you can change your code temporarily to be 

```
    [MonitoredItem("ns=27;s=NC.SystemData,/MachineData/ServiceLocation")]
    public ExtensionObject ServiceLocation{
        get => serviceLocation;
        set => SetProperty(ref serviceLocation, value);  // set breakpoint here, and read the value's TypeId
    }
    private ExtensionObject serviceLocation;

```

Set a Breakpoint to stop when the property is written.  The type ExtensionObject has a field TypeId that holds the NodeId we are after.  After you have read the TypeId,  you can return your code to the original.

Lastly, you will have to register the custom type with the library. 

```
var app = new UaApplicationBuilder()
  <snip>
  .AddTypes(typeof(Cartesian))
  .Build();

```

invinctus, 2020-10-20:
I have  .AddTypes(typeof(Cartesian)) in my Application Builder.
    [DataTypeId("ns=28;s=Cartesian_ty")]
    [BinaryEncodingId("nsu=http://boschrexroth.com/OpcUa/CNC/Encoding/Types/;s=Cartesian_ty")]
    public class Cartesian : Structure

For my Structure declaration

Now I get a BadDecodingError

quinmars, 2020-10-20:
In your original example, you are missing the `override` modifier for the `Encode` and `Decode` method. Without that, only the base implementation of `Structure` will be called.

invinctus, 2020-10-20:
I see it! I started by just implementing the interface then I spotted the structure class and totally forgot to add the override modifiers.

It's working great now thank you very much for all the help :)

closed 2020-10-20 by invinctus