convertersystems/opc-ua-client #175: FindServersOnNetwork

ismiwo, 2020-11-21:
I can't find a method to make use of the FindServersOnNetworkRequest.
Shouldn't there be a method inside UaTcpDiscoveryService for this purpose?

quinmars, 2020-11-21:
Good question! As I understand the [specification](https://reference.opcfoundation.org/v104/Core/docs/Part4/5.4.3/), the _FindServersOnNetwork_ request can be used to paginate the results. Hence one probably want to keep the channel open, so it can be reused in the follow up queries. On the other hand I like the simpliness of the current `FindServersAsync` and `GetEndpointsAsync` static methods, where you do not need to care about the creation and disposal of the channel. What do you think, should `FindServersOnNetworkAsync` be an instance method or should it be static?

awcullen, 2020-11-22:
@ismiwo Which server do you use?  The demo server from Unified automation does not seem to support this service request.

ismiwo, 2020-11-22:
I'm using devices in our network that have built-in opc-ua-servers and I am using the server from this project: [nauful/LibUA](https://github.com/nauful/LibUA)

Thank you for pointing to the specification. I found in [4.3.5](https://reference.opcfoundation.org/v104/GDS/docs/4.3.5/) that a client can call this function on a LDS with MulticastExtension.
I also found that servers don't have to register themselves ([4.2.1](https://reference.opcfoundation.org/v104/GDS/docs/4.2.1/)).

I thought that UaExpert was able to discover servers automatically. But it was always only discovering my local server which doesn't seem to have LDS-ME capabilities. But UaExpert is also trying FindServersOnNetwork first when you want to add a server:

[image]

I realize now that I cannot rely on all the servers to be discovered automatically and that I will have to add other measures to discover them in our software (e.g. [4.3.1](https://reference.opcfoundation.org/v104/GDS/docs/4.3.1/)).

@quinmars I think the method should be static.

quinmars, 2020-11-30:
An LDS-ME can be downloaded here: https://opcfoundation.github.io/UA-LDS/ - I haven't tried it yet.