convertersystems/opc-ua-client #182: Can we have a SubscriptionBase without SubscriptionAttribute?

ismdiego, 2021-01-21:
First of all, awesome library! I have only one question. In order to easily use this library, I created some ViewModels with the SubscriptionAttribute.

All is going well and so, but I would like to have other variant of SubscriptionBase that accepts the subscription attribute data at runtime. The problem is that sometimes the endpoint URL changes and this forces us to recompile the project with the new attribute set (as it must be a constant). It would be great to be able to pass this information at runtime (previously read from a configuration file).

Maybe I am missing something obvious, but I don't find how to achieve this. I tried copying the code in a new "SubscriptionBaseEx" in my assembly but it uses UaApplication.Completion which is internal. I could change that in the source, but I don't know the possible implications, so better safe than sorry.

Anyway, the purpose of this is to have a very simple ViewModel (like in your sample), weaved with Fody to implement the PropertyChanged automatically. This works brilliant in this use case. The only "but..." point is the required SubscriptionAttribute. Simply adding a new constructor will do the trick for my use case.

I can create a Pull request if you want. Just tell me.

Thanks and congratulations again for such a great library.

awcullen, 2021-01-24:
Hi,
I'd definitely like to see your pull request.  
You can also try adding an option "AddMappedEndpoint" to the UaApplicationBuilder
The idea is that you map the Url in the SubscriptionAttribute to a new EndpointDescription where you can specify a different Url and SecurityPolicy.  

 
```
var app = new UaApplicationBuilder()
    ...
    .AddMappedEndpoint("opc.tcp://localhost:48010", new EndpointDescription {EndpointUrl = "opc.tcp://192.168.1.100:48010"})
    .Build();

[Subscription(endpointUrl: "opc.tcp://localhost:48010", publishingInterval: 500, keepAliveCount: 20)]
private class MyViewModel : SubscriptionBase
{}
``` 

Further, if you don't wish to recompile you can use a config file to do the same.

```
using Microsoft.Extensions.Configuration;

var config = new ConfigurationBuilder()
    .SetBasePath(Directory.GetCurrentDirectory())
    .AddJsonFile("appSettings.json", true)
    .Build();

var app = new UaApplicationBuilder()
    ...
    .AddMappedEndpoints(config)
    .Build();

[Subscription(endpointUrl: "ProductionServer", publishingInterval: 500, keepAliveCount: 20)]
private class MyViewModel : SubscriptionBase
{}
```
appSettings.json
```
{
  "MappedEndpoints": [
    {
      "RequestedUrl": "WarehouseServer",
      "Endpoint": {
        "EndpointUrl": "opc.tcp://localhost:49320",
        "SecurityPolicyUri": "http://opcfoundation.org/UA/SecurityPolicy#None"
      }
    },
    {
      "RequestedUrl": "ProductionServer",
      "Endpoint": {
        "EndpointUrl": "opc.tcp://localhost:48010",
        "SecurityPolicyUri": "http://opcfoundation.org/UA/SecurityPolicy#None"
      }
    }
  ]
}
```

ismdiego, 2021-01-24:
Oh! I did not realize about the MappendEndpoint feature, but it's really great! Thank you so much @awcullen for your help.

I prefer this over my proposed change, but anyway I have created a pull request with my *trivial* changes. Also, as I found this information you gave me so important, I also included in the pull request modifications to the README.md file adding your comments.

Feel free to cherry pick whatever you want. I think that at least the README.md modifications should be included in the main branch, to help other developers having these runtime requirements like me.

One last question about the VM usage. When I dispose the VM, should I notify somewhat the SubscriptionBase class for it to stop the inner OPCUA subscription? I have some fear of causing memory leaks if not doing it well.

Thank you very much again

closed 2021-03-06 by awcullen