convertersystems/opc-ua-client #185: WPF on .NET 5

dubeg, 2021-02-06:
Hi team,
I just started playing with your library, and so I created a Console on .NET 5 using your sample code: everything worked fine. However, I couldn't get it to work on Wpf on .NET 5. I set it up exactly as in your sample Wpf .NET 4.6 app, but it wouldn't log anything in the Output window and the window would only show "connecting". I don't know much about async machinery, but I noticed it hanged on await this.whenSubscribed.Task in SubscriptionBase.

On the other hand, when I recreated your wpf sample using .NET 4.8, everything worked fine.

Is this something you ever tried?

dubeg, 2021-02-06:
Wait, my mistake. My working sample had set `DataContext = MainViewModel`, while the non-working sample had MainViewModel as a property on MainWindow, and even though the bindings were like `Binding ViewModel.ServerServerStatus`, that's what made it break.

closed 2021-02-06 by dubeg

awcullen, 2021-02-07:
Thanks for testing the library.  We appreciate your feedback!