convertersystems/opc-ua-client #186: Read and Write Model

franklupo, 2021-02-08:
Hi,
you did a wonderful job. I can't find a way to write and read a model class.
How can I do?

best regards

awcullen, 2021-02-09:
Hi,
If I am understanding correctly that your model is a custom type, then there a couple of steps.

First, I recommend you create a  custom TypeLibrary.  See https://github.com/convertersystems/opc-ua-client/blob/568ee1339a43ff4906eeb85400d2b1b5c3b7e693/CustomTypeLibrary/CustomVector.cs#L5
The types in this library must subclass Structure and provide an Encode and Decode method.  They also need some attributes including the BinaryEncodingId that matches the id the server is expecting.

Second, you will be able to read the custom structure just as easy as reading the ServerStatus as shown in this example:
https://github.com/convertersystems/opc-ua-client/blob/568ee1339a43ff4906eeb85400d2b1b5c3b7e693/UaClient.UnitTests/IntegrationTests/IntegrationTests.cs#L283

franklupo, 2021-02-09:
Hi,
similar SubscriptionBase a class and properties with attributes.

Best regards

closed 2021-02-14 by franklupo