convertersystems/opc-ua-client #187: OPC - UA reading alarms client

harishLinkEZ, 2021-02-19:
We can add OPC - UA alarms client  to read alarms

awcullen, 2021-02-21:
Yes,  I use this UA client to subscribe to events and alarms, and to acknowledge the alarms. 

A short example that works with the free UaCPPServer from Unified Automation

https://github.com/convertersystems/OpcUaFeatureTests/blob/c6c3ed931047d861fa5a5ade730a9937e3532967/OpcUaFeatureTests/SubscriptionTests.cs#L125

harishLinkEZ, 2021-02-24:
Thank you Expert. I will check this out.

harishLinkEZ, 2021-03-24:
I was trying to read alarms presents in my Siemens 828D , First I tried with UAExpert tool, through which I could read the alarms.
Then I tried the mentioned code.

Connection -> Success
Authentication -> Success
Event Subscription -> Success
reading alarms -> failure or no success
Am I doing anything wrong here. I have modified few parameters for connecting to Siemens HMI. Thank you very much for your time.

`        public static async Task CreateEventSubscription()
        {
            // describe this client application.
            var discoveryUrl = $"opc.tcp://" + "192.168.1.1 + ":" + 4840;

            Console.WriteLine("Step 1 - Describe this app.");
            var appDescription = new ApplicationDescription()
            {
                ApplicationName = "OPCWorkstation",
                ApplicationUri = $"urn:{System.Net.Dns.GetHostName()}:OPCWorkstation",
                ApplicationType = ApplicationType.Client,
            };

            Console.WriteLine("Step 2 - Create a certificate store.");
            var certificateStore = new DirectoryStore(
                Path.Combine(Environment.GetFolderPath(Environment.SpecialFolder.LocalApplicationData), "OPCWorkstation.ConsoleApp", "pki"));//Workstation.UaClient.

            Console.WriteLine("Step 3 - Create a session with your server.");
            var channel = new UaTcpSessionChannel(
                appDescription,
                certificateStore,
                new UserNameIdentity("OpcUaClient", "OpcUaClient"),
                discoveryUrl,
                SecurityPolicyUris.Basic128Rsa15);

            try
            {
                await channel.OpenAsync();

                Console.WriteLine($"  Opened channel with endpoint '{channel.RemoteEndpoint.EndpointUrl}'.");
                Console.WriteLine($"  SecurityPolicyUri: '{channel.RemoteEndpoint.SecurityPolicyUri}'.");
                Console.WriteLine($"  SecurityMode: '{channel.RemoteEndpoint.SecurityMode}'.");
                Console.WriteLine($"  UserIdentity: '{channel.UserIdentity}'.");
                Console.WriteLine($"Opened session with endpoint '{channel.RemoteEndpoint.EndpointUrl}'.");
                Console.WriteLine($"SecurityPolicy: '{channel.RemoteEndpoint.SecurityPolicyUri}'.");
                Console.WriteLine($"SecurityMode: '{channel.RemoteEndpoint.SecurityMode}'.");
                Console.WriteLine($"UserIdentityToken: '{channel.UserIdentity}'.");

                // build a CreateSubscriptionRequest. See 'OPC UA Spec Part 4' paragraph 5.13.2
                var req = new CreateSubscriptionRequest
                {
                    RequestedPublishingInterval = 500.0, // intervals are in milliseconds
                    RequestedMaxKeepAliveCount = 30,
                    RequestedLifetimeCount = 30 * 3,
                    PublishingEnabled = true,
                };
                var res = await channel.CreateSubscriptionAsync(req);

                // the result will return the server's subscription id. You will needs this to 
                // add monitored items.
                var id = res.SubscriptionId;
                Console.WriteLine($"Created subscription '{id}'.");

                NodeId eventNodeId;
                NodeId.TryParse("ns=2;s=Sinumeric", out eventNodeId);
                
                // build a CreateMonitoredItemsRequest. See 'OPC UA Spec Part 4' paragraph 5.12.2
                var req2 = new CreateMonitoredItemsRequest
                {
                    SubscriptionId = id,
                    TimestampsToReturn = TimestampsToReturn.Both,
[... 36 lines cut ...]
                            {
                                foreach (var efl in enl.Events)
                                {
                                    var ev = EventHelper.Deserialize<BaseEvent>(efl.EventFields);
                                    Console.WriteLine($"time: {ev.Time}, src: {ev.SourceName}, msg: {ev.Message}, sev: {ev.Severity}");
                                }
                            }
                        },
                        ex => Console.WriteLine("Exception in publish response handler: {0}", ex.GetBaseException().Message)
                   );
                while(true)
                {
                    Task.Delay(500);
                }
                await Task.Delay(500);
            }
            catch (Exception ex)
            {
                await channel.AbortAsync();
                Console.WriteLine(ex.Message);
            }
        }
      
    }
`

harishLinkEZ, 2021-03-30:
You have to subscribe to the node given below: 
            NodeId eventNodeId;
            NodeId.TryParse("ns=2;s=Sinumerik", out eventNodeId);
            and you have to give access to the user " Sinumeric/Methods/GiveUserAccess" using your username & "SinuReadAll", Find the details how to give user access to events in following manual:
https://cache.industry.siemens.com/dl/files/643/109772643/att_1001368/v1/840Dsl_828D_OPCUA_config_man_0819_en-US.pdf
under alarms&events

harishLinkEZ, 2021-03-30:
Thank you, let me know if you guys need anything

closed 2021-03-30 by harishLinkEZ