convertersystems/opc-ua-client #190: Guidance for use in ViewModels (Subscription not reading)

ismdiego, 2021-02-25:
I am using this package with the ViewModel [Subscription] attribute, but sometimes I have serious problem to make it work.

I suspect this has to be related to the fact that a Task is really started and then the subscription begins. In some of my view models I can read and write properties with no problem, and in some others I cannot even read a OPCUA tag. The value of the property never updates.

Is there some guidance for this use case you can share?

About the details, I am using this in a Windows Forms desktop application (not WPF) and I initialize the viewmodel just before the Application.Run(...) code. Maybe this is not appropriate, but I do this for all the viewmodels and some work and other does not.

I am still testing to try to isolate and reproduce my problem in a simple scenario, but I would like to know if someone more is affected by this behavior. I also use Reactive extensions together with ReactiveUI, and so I tried to ObserveOn in the main thread with no luck (although this really helps when writing properties).

Many thanks

awcullen, 2021-02-26:
You will need to construct the UaApplication first, and then construct the viewmodels.

I have not used System.Reactive with WinForms.  I seems to be supported.

Let me know if this solves your issue.

closed 2021-03-06 by awcullen

ismdiego, 2021-03-08:
Sorry, I could not test this with a test project, but soon I will be able to do it and I will write my comments here.

About your suggestion, yes, the UaApplication is built and started before any ViewModel exists. And I currently have 3 viewmodels in one project working and 1 not communicating. I checked the code but it seems that it's the same. Obviously I am doing something wrong, but I cannot determine what yet.

And yes, Reactive is supported in Windows Forms, in fact I am using ReactiveUI (a MVVM framework built on top of Reactive) and it works perfectly. And, as I said, I already have 3 working ViewModels with OPC UA subscription running well.

The other problem that I would like to clarify is how the subscription is closed. Because sometimes, after several tests, I get a "maximum number of subscriptions error" in the OPC UA Server I'm using. And it has to be related to them not closing well. I am doing tests and constantly shutdown the application with the stop button in Visual Studio, but I fear that this can happen in production...

awcullen, 2021-03-10:
Great to hear that Windows Forms is working!

If the client app exits without cleaning up the subscriptions, the UA Server will do it after time not less than lifetimeCount. 
By default, I set the lifetimeCount at 3 times the keepAliveCount. 

So the following would be cleaned up after 30 seconds.
[Subscription(endpointUrl: "opc.tcp://localhost:48010", publishingInterval: 500, keepAliveCount: 20)]

ismdiego, 2021-06-10:
@awcullen , regarding the cleaning of the subscriptions. In my case the viewmodels are used with ReactiveUI MVVM framework to update bindings on screen. It happens that I used a "WhenAnyValue" method to subscribe for change events in VM properties but forgot to unsubscribe when disposing it... and so the GC could not close that relation.

Analyzing the SubscriptionBase code, I see that in PropertyChangedEventHandler event, at the remove part, it checks the invocation list to see if it is only 1, and then (and only then) tries to set true to whenUnsubscribed TaskCompletionSource<bool>. In turn, I found this the *only* way to stop the subscription, because in the StateMachineAsync you can see:

`// wait here until channel is closing, unsubscribed or token cancelled.`
`try`
`{`
`    await Task.WhenAny(`
`    this.WhenChannelClosingAsync(this.innerChannel, token),`
`    this.whenUnsubscribed.Task);`
`}`

As the CancellationToken (token) is private (not protected) one cannot stop the subscription that way, making useless (I think) the first condition. So the only way to close the subscription for me was to correctly disposing the VM property change handlers.

This way I could "fix" the "Server has reached maximum number of subscriptions" error when using this library after a while...

Maybe this message can help others having the same problem