convertersystems/opc-ua-client #195: Linq to OPC query service

quinmars, 2021-03-17:
UaClient provides access to the plain query service set. The query service set is not very easy to use, at least for me who has never used it. 😄  I think it'd be really great if we would have support for `IAsyncQueryable` to ease the use of this service. So querying all nodes that have a node class of `DataType`:

```csharp
var allDataTypes = channel.Nodes.Where(x => x.NodeClass == NodeClass.DataType);

await foreach (var datatype in allDataTypes)
{
    // ...
}
```

`Nodes` or another fitting name would return an `IAsyncQueryable` and the `Where`-clause will be evaluated, due to the underlying mechanics, on the OPC server. This is not a finalized proposal. I know that this will not be simple to implement and there are probably many LINQ operators that cannot be translate into a query request, but I think that would be really great to support as much as possible.

dojo90, 2021-03-18:
If that were possible, that would be awesome! 😃👍
If you need help at this point for further implementation, I am happy to support you.

I am currently searching the entire OPC server to instantiate my software accordingly. That would surely save me a lot of time!

quinmars, 2021-03-22:
> If you need help at this point for further implementation, I am happy to support you.

Thanks, at the moment I'm still familiarize myself with the query service and C#'s query expressions. But once I have something to show, I'll let you know.

quinmars, 2021-04-15:
Here is a short update on my progress on that issue. I have good and bad news.

Let's start with the good news. I studied the OPC UA query service set and C# query expressions. The query service set is indeed very powerful. I think it is possible to support the following operators: `Select`, `Where`, `Take`, `TypeOf`. Combining operators like `SelectMany`, `Join`, `Zip` and aggregating operators like `Min`, `Max`, etc. are probabliy not possible. Due to the filter abilities of the query service set, the `Where` operator could support very complex expressions, which are server-side evaluated. I already started with the implementation.

But here comes the bad news. While playing with the query service set, I recognized that the `UaCPPServer` apparently does not support the query service set. After some researched I found the following paper: [Querying OPC UA information models with SPARQL](https://ieeexplore.ieee.org/document/8868246). There the authors state:

> The OPC UA Query ServiceSet is probably the only OPC UA ServiceSet which has no publicly available implementation up to now. Even the fact that a whole annex was introduced in OPC UA Part 4, which added a lot of examples for this Service, has not fulfilled its goal.

That curbed my enthusiasm a bit. Without a server supporting the query service set, it is more like an academic finger exercise. Nonetheless, it is possible to implement the query methods with unit tests only.

closed 2021-06-15 by awcullen