convertersystems/opc-ua-client #198: Configure EndpointUrl and nodeId at runtime (MVVM)

dejanbatljan, 2021-04-22:
I need to create generic client application (WPF with MVVM) where only endpointUrl and nodeId parameters in MonitoredItem attributes would need to be different at runtime.

I can see, that endpointUrl can be replaced at runtime with 
.AddMappedEndpoints(config)

I need advice how to do similar thing with nodeId parameter in MonitoredItem attribute in viewmodels.

Since attribute parameters need to be evaluated at compile time, I'm having troubles defining them dynamically.

Any advice how to approach this?

awcullen, 2021-04-24:
I did not anticipate a need to change the NodeID at runtime.

If you are writing an OPC UA Browser type application you will find this project very interesting.
https://github.com/quinmars/UaLayman

quinmars, 2021-04-26:
That reminds me to put some more work into the UaLayman. I still want to publish it in the store at some day. I probably should hurry up before MS is closing the store.

closed 2021-05-12 by awcullen