convertersystems/opc-ua-client #200: Encodable Type Problem

invinctus, 2021-05-13:
I am struggling to map a node to my viewmodel.

This is what the node looks like in UaExpert
[image]

It is an array of "AxisPositionDataType", first of all i added it as an object to get the binary decoder id.  Then I created a class to match the AxisPositiionDataType
`   [BinaryEncodingId("nsu=http://boschrexroth.com/OpcUa/CNC/Motion/Types/;i=7022")]
    public class AxisPositionData : Structure
    {
        public double Position { get; set; }
        public string Name { get; set; }
        public bool InPosition { get; set; }
        public override void Decode(IDecoder decoder)
        {
            Position = decoder.ReadDouble(nameof(Position));
            Name = decoder.ReadString(nameof(Name));
            InPosition = decoder.ReadBoolean(nameof(InPosition));
        }

        public override void Encode(IEncoder encoder)
        {
            throw new NotImplementedException();
        }
    }`

now I updated my view model to map the type to this class:
`        [MonitoredItem("ns=27;s=NC.Chan.AxisPosAcsDrv,01")]
        public AxisPositionData[] AcsAxisData { get => acsAxisData; set => SetProperty(ref acsAxisData, value); }`

But this gives me an error: "Error publishing value for NodeId ns=27;s=NC.Chan.AxisPosAcsDrv,01. Object of type 'System.Object[]' cannot be converted to type 'AxisPositionData[]'."

I can see that it is decoding the array elements correctly but I cant figure out how to get this to work

awcullen, 2021-05-17:
A workaround for the moment would be to implement two properties.  The first uses object{] as the property type. The second casts the object[] to the AxisPositionData[]

```
[MonitoredItem("ns=27;s=NC.Chan.AxisPosAcsDrv,01")]
public object[] AcsAxisData { get => acsAxisData; set => SetProperty(ref acsAxisData, value); }

public AxisPositionData[] AcsAxisData2 { get => AcsAxisData.Cast<AxisPositionData>().ToArray(); }

```

I'll let you know when I've got something better :)

awcullen, 2021-05-29:
I have released a new version 3.1.0 that supports a view model property as you originally wrote.

```
[MonitoredItem("ns=27;s=NC.Chan.AxisPosAcsDrv,01")]
public AxisPositionData[] AcsAxisData { get => acsAxisData; set => SetProperty(ref acsAxisData, value); }
```
Thank you for reporting the issue.

invinctus, 2021-05-29:
Thats great I'll get that implemented.

closed 2021-06-06 by awcullen

invinctus, 2021-08-16:
Sorry it has taken me so long to get back around to this but I updated to package version 3.1.0 and I am still getting the same error
"Error publishing value for NodeId ns=27;s=NC.Chan.AxisPosAcsDrv,01. Object of type 'System.Object[]' cannot be converted to type 'AxisPositionData[]'."