convertersystems/opc-ua-client #203: Is there a way to read the NodeId and Value of a Node simultaneously?

jaarfi, 2021-06-09:
Hello everyone,

I am using your library for a project, where I want to subscribe to multiple Nodes in a single Subscription. Putting multiple Nodes into a single subscription works well, however after receiving the PublishResponse, I am not sure how to  know which values belongs to which Node. Is there a way to find out what the NodeIds used for the MonitoredItems are?

dojo90, 2021-06-10:
Hi @jaarfi,
you can have a look at my implementation of a `SubscriptionManager`: https://gist.github.com/dojo90/a660b5181d980c2a070b1b730aba733c

awcullen, 2021-06-10:
Good question!

When you call CreateMonitoredItems(), you can assign a unique integer to the ClientHandle field of each MonitoredItem.
When you receive the PublishResponse, the MonitoredItemNotification will contain the ClientHandle field which tells you which value you received.   Some people keep a Dictionary of Nodes that use the ClientHandle as an index.

Hope this helps

closed 2021-06-15 by awcullen