convertersystems/opc-ua-client #204: Incorrect or empty ServerTimestamp?

ShuhuaGao, 2021-06-22:
Hi,

After reading a node, I inspected the `ServerTimestamp` and the `SourceTimestamp`. However, the `ServerTimestamp` seems to always be "0001/1/1 0:00:00".

- Read a node that simply yields random number
```cs
var randDataValue = readResult.Results[2];
Console.WriteLine("\nRandom node: ");
Console.WriteLine($"    Status: {randDataValue.StatusCode}");
Console.WriteLine($"    Value: {randDataValue.GetValueOrDefault<double>()}");
Console.WriteLine($"    Server timestamp: {randDataValue.ServerTimestamp}");
Console.WriteLine($"    Source timestamp: {randDataValue.SourceTimestamp}");
```
- The result is:
[image]

- In the simulation server end, both timestamps are correct.
[image]

Thanks for any help.

awcullen, 2021-06-22:
Hi,

You can get both timestamps by setting a property of the CreateMonitoredItemsRequest.  

Example:
```
    var itemsRequest = new CreateMonitoredItemsRequest
    {
        SubscriptionId = id,
        ItemsToCreate = itemsToCreate,
        TimestampsToReturn = TimestampsToReturn.Both
    };

```

Best Regards,
Andrew

ShuhuaGao, 2021-06-23:
Hi, Andrew. Many thanks.

By the way, could I ask you another question, since I am totally new to OPCUA? My application needs to read node information via OPCUA every 2 minutes. In that case, shall I keep the connection (i.e., a channel) open for the whole application's lifetime? Or should I close the channel once finishing the reading and then open it again next time (after 2 minutes)? Which one is the best practice in your opinion?

closed 2021-07-03 by ShuhuaGao