convertersystems/opc-ua-client #209: Specify data type when building ReadValueId

easysoft2k15, 2021-08-02:
Is there a way to specify the data type when building the ReadValueId?
I'm using the Kepserver OPC with the GE Fanuc ethernet drive.
The code is as follow:

```
            var itemToCreate = new MonitoredItemCreateRequest
            {
                ItemToMonitor = new ReadValueId {
                                        NodeId = NodeId.Parse($"ns=2;s=${variable}"),
                                        AttributeId = AttributeIds.Value
                                        },
                MonitoringMode = MonitoringMode.Reporting,
                RequestedParameters = new MonitoringParameters
                {
                    ClientHandle = 1,
                    SamplingInterval = -1,
                    QueueSize = 0,
                    DiscardOldest = true
                }
            };
```

variable is boolean.
There are 2 cases:
1. WORKING CASE:
variable is defined inside the OPC server with boolean type and the definition inside C# is as follow:
`variable=channel1.device1.var1`

2. FAILING CASE:
variable is NOT defined inside the OPC server and the definition inside C# is as follow:
`variable=channel1.device1.!var1`
Note the 

> **!**

 character preceding the var1 which allow a definition by address of the variable (variable is declared inside the PLC as var1 but is not declared on the Kepware OPC server)

If I read a short type variable both cases works thus I assume that is a data type problem.
Any help would be appreciated.
Thank You

awcullen, 2021-08-05:
While there is no official OPC way to specify the DataType when building the MonitoredItemCreateRequest, there just might be a Kepware server specific syntax to do so.  The syntax would have to be passed in the NodeID, just like the '!' character you found.

easysoft2k15, 2021-08-06:
I investigated more and You are right!
The NodeId  syntax allow indeed to specify data type as follow:

> [Tag Name]@[Data Type]

for example:

> ns=2;s=channel1.device1.var1@Float

For anyone interested please check the following Keepware knowledge base:
[static-and-dynamic-tags](https://www.kepware.com/en-us/support/knowledge-base/2008/static-and-dynamic-tags/)
Thank You!

closed 2021-08-06 by easysoft2k15