convertersystems/opc-ua-client #211: Unable to create NodeId -

avnet78, 2021-08-14:
I am trying to read the nodes under Real Time Data, there are about 20 tags. I am able to create the session without any issues but for some reason I am unable to create the NodeId. 
As you can see in the code snipped below, I tried both NodeId.Parse and new Node(identifer, namespaceindex), but both are throwing an exception.

Please help!

Here is my code:

`var opcTags = new List<string> { "Tag1", "Tag2", "Tag3", "Tag4", "Tag5", "Tag6", "Tag7", "Tag8", "Tag9", "Tag10", "Tag11", "Tag12", "Tag13", "Tag14", "Tag15", "Tag16", "Tag17", "Tag18", "Tag19", "Tag20" };
var readValueIds = new List<ReadValueId>();

foreach (var item in opcTags)
{
    var tempId = $"ns=2;s={item}";

    var nodeId = NodeId.Parse(tempId);
    //new NodeId(item, 2);

    var readVal = new ReadValueId
    {
        NodeId = nodeId,
        AttributeId = AttributeIds.Value
    };

    readValueIds.Add(readVal);
}

// build a ReadRequest. See 'OPC UA Spec Part 4' paragraph 5.10.2
var readRequest = new ReadRequest
{
    // set the NodesToRead to an array of ReadValueIds.
    NodesToRead = readValueIds.ToArray()
};
// send the ReadRequest to the server.
var readResult = await channel.ReadAsync(readRequest);`

[image]

[image]

quinmars, 2021-08-14:
It looks like you are missing the `Workstation.UaClient.dll`. Usually visual studio copy it to your executable folder. You may want to check if it is actually present.

awcullen, 2021-08-14:
It looks like QuickWatch window is having problem finding the NodeId type.  The application is finding the dll just fine, because it opened the session channel with server properly.

More importantly, I want to point out that the Ua Browser is showing the DisplayName of each node in the "Real Time Data" folder.  Often the NodeId is different from the DisplayName of a Node. The NodeId may be a String or Int (or sometimes a GUID or byte array).  All NodeIds must be unique in their namespace. You may find the server vendors use long strings in the NodeId to ensure they will be all unique, like "ns=3;s=AirConditioner_1.Temperature.EngineeringUnits".  

I would use a free tool such as UaExpert from UnifiedAutomation to discover the proper NodeId of the Nodes you wish to read.

avnet78, 2021-08-14:
Installed UnifiedAutomation's UaExpert as suggested. I think the node id is "ns=2;s=Tag1". I am using the same node id in my code. Please advise!

[image]

avnet78, 2021-08-14:
Ok, I tried the old trick - ran the disk cleanup and restarted my machine. I can read the values now.
Some things don't change with Microsoft.  ;)

[image]

avnet78, 2021-08-15:
Finally I was able to resolve the issue. 
I am not really sure but adding the below namespace dramatically fixed the issue. I was looking at your code samples realized that I didn't have this in my code. 
I am able to read the values for the OPC UA Server simulator. 

````using System.Reactive.Linq;````

[image]

closed 2021-08-15 by avnet78