convertersystems/opc-ua-client #214: Subscriber exception handling

DevAlascom, 2021-11-15:
I developed a subscriber to get latest updated values from server. It works well until I disconnect the Lan Ethernet cable. Then application crashed and I cannot handle the exception.

Could you please let me know how to handle this condition?

`var channel = UaTcpSessionChannelInstance();
            try
            {
                if (channel != null)
                {
                    await channel.OpenAsync();

                    Console.WriteLine($"  Opened channel with endpoint '{channel.RemoteEndpoint.EndpointUrl}'.");
                    Console.WriteLine($"  SecurityPolicyUri: '{channel.RemoteEndpoint.SecurityPolicyUri}'.");
                    Console.WriteLine($"  SecurityMode: '{channel.RemoteEndpoint.SecurityMode}'.");
                    Console.WriteLine($"  UserIdentity: '{channel.UserIdentity}'.");

                    var subscriptionRequest = new CreateSubscriptionRequest
                    {
                        RequestedPublishingInterval = 500,
                        RequestedMaxKeepAliveCount = 100,
                        RequestedLifetimeCount = 30,
                        PublishingEnabled = true
                    };

                    var subscriptionResponse = await channel.CreateSubscriptionAsync(subscriptionRequest);
                    var id = subscriptionResponse.SubscriptionId;

                    Console.WriteLine("Step 6 - Add items to the subscription.");
                    var itemsToCreate = InitiateTagsToSubscribe();
                    var itemsRequest = new CreateMonitoredItemsRequest
                    {
                        SubscriptionId = id,
                        ItemsToCreate = itemsToCreate,
                    };
                    DataBaseHandler.InitiateTags(MobotController.mobots.Count);
                    var itemsResponse = await channel.CreateMonitoredItemsAsync(itemsRequest);

                    var token = channel.Where(pr => pr.SubscriptionId == id).Subscribe(pr =>
                    {
                        isConnected = true;
                        // loop thru all the data change notifications
                        var dcns = pr.NotificationMessage.NotificationData.OfType<DataChangeNotification>();
                        var _tags = ApplicationConfiguration.Current.appSetting.opcuaConfig.Tag;
                        foreach (var dcn in dcns)
                        {
                            foreach (var min in dcn.MonitoredItems)
                            {
                                Console.WriteLine($"sub: {pr.SubscriptionId}; handle: {min.ClientHandle}; value: {min.Value}");

                            }
                        }
                    });
                }
                else
                {
                    isConnected = false;
                }
            }
            catch (AggregateException ea)
            {
                isConnected = false;
                Console.WriteLine("Error in SubscribeListenToPlc: " + ea.Message);
                channel.CloseAsync();
                await channel.AbortAsync();
                //LogController.WriteErrorLog(LogController.ExceptionMessageParser(e, MethodBase.GetCurrentMethod()));
                SubscribeListenToPlc();
            }
            catch (Exception e)
            {
                isConnected = false;
                Console.WriteLine("Error in SubscribeListenToPlc: " + e.Message);
                channel.CloseAsync();
                await channel.AbortAsync();
                //LogController.WriteErrorLog(LogController.ExceptionMessageParser(e, MethodBase.GetCurrentMethod()));
                SubscribeListenToPlc();
            }`

awcullen, 2021-11-18:
I believe this may handle it:
```
    var token = channel.Where(pr => pr.SubscriptionId == id).Subscribe(pr =>
    {
        isConnected = true;
        // loop thru all the data change notifications
        var dcns = pr.NotificationMessage.NotificationData.OfType<DataChangeNotification>();
        var _tags = ApplicationConfiguration.Current.appSetting.opcuaConfig.Tag;
        foreach (var dcn in dcns)
        {
            foreach (var min in dcn.MonitoredItems)
            {
                Console.WriteLine($"sub: {pr.SubscriptionId}; handle: {min.ClientHandle}; value: {min.Value}");

            }
        }
    },
    ex =>
    {
        Console.WriteLine($"Exception in publish response handler: {ex.GetBaseException().Message}");
    });
```

DevAlascom, 2021-12-16:
Thank you so much.
I have another question. I would like to know your opinion about read and write.
I need to keep the session alive for ever, Do I need to create 2 seperate session for read and write or I can create one and use it for both?

awcullen, 2021-12-18:
I often use one session for both reading and writing.  Your program will need to share the open channel between the reading task and the writing task.  Sometimes this can be tricky in async programs. 

If the writing is performed rarely, then you may be better off creating a second session, write the tag, and then close the second session.

closed 2023-03-16 by awcullen