convertersystems/opc-ua-client #217: Missing notification

LouisBaranger, 2021-12-14:
Hello,

I'm working with your client with a Siemens CPU 1500 server. 

It works well except sometimes, quite randomly, I don't get the notification from some changes on monitored items.
I have watched on PLC, the item has changed of value.
I don't have any exceptions, neither notification for this new value. 
I used Wireshark to see what append on Tcp and I don't see any dataChange for this item.

If I manualy change the value few seconds later, I'm getting a notification...

Some details about my sub and items.
Subscriptions : 
Publish interval : 500,
Lifetime count : 600,
MaxKeepAlive : 60

MonitoredItems :
Sampling interval : 500
Queue Size : 10
DiscardOldest : True

This problem is blocking my program because I wait the end of a process with a subscription on PLC but sometimes, I never get it.

Thanks,

PS :  I already have watched the bkohlmaier issue, I think this is the same problem.

awcullen, 2021-12-16:
Hi Louis,

I would like to learn more about your program, for example: subscription count, items per subscription.  You mentioned you have experience with Wireshark. Could you share some details about the returned values from CreateSubscriptionResponse and CreateMonitoredItemsResponse?
 
Also, I am curious if you can share the PublishResponse. If the server sends a PublishResponse and the client doesn't acknowledge it. There will be a growing list of AvailableSequenceNumbers.  

If you are on a tight schedule with your project, maybe you can program a periodic read of this Item and determine the end of your process that way.   

I do hope to find the source of this bug since both you and bkohlmaier have reported it.

Andrew

LouisBaranger, 2021-12-16:
Thanks for your interest,

It's a variable application, in this case, I use 1 session, 1 sub, 68 monitored items, it can be in the worse case around 200 monitored items. I tested to read all this items (200), it takes me around 1200ms...

I have checked the AvailableSequenceNumbers in publish around the bug, the array size is always 0.

Here you can see some capture from Wireshark. 
GitHub doesn't support .pcapng file, if you give me an email I can send you all the detail.

Channel response
[image]

Session Response
[image]

Subscription Response
[image]

MonitoredItem Response
[image]

Publish Request
[image]

Publish Response
[image]

awcullen, 2021-12-18:
Thanks for all the good info. 

I setup a test using PLCSim with a tag that increments every 0.1 s.  The ua server settings are:

[image]

I wrote a console app to log the data change notifications. The item is set to sample at 1s and the subscription is set to publish at the same rate.
I also tested sampling at 1s and publishing at 2s.  In this case, I get PublishResponse every 2s, but the response has two notifications in it.  

An alternative to subscription is to read the list of nodes.  You may notice that reading can be slower.  To speed up reading you pass the list of nodes to the RegisterNodes method.  The response gives you a list of nodeID that use to read instead of the original.  [Let me know if you would like an example]
  
Info about RegisterNodes  is here: https://reference.opcfoundation.org/v104/Core/docs/Part4/5.8.5/

awcullen, 2021-12-18:
I found an example 
```
        private static async Task TestAsync2()
        {

            // var discoveryUrl = "opc.tcp://localhost:48010"; // UaCppServer - see  http://www.unified-automation.com/
            var discoveryUrl = $"opc.tcp://192.168.1.94:4840"; // Siemens

            Console.WriteLine("Step 1 - Describe this app.");
            var appDescription = new ApplicationDescription()
            {
                ApplicationName = "MyHomework",
                ApplicationUri = $"urn:{System.Net.Dns.GetHostName()}:MyHomework",
                ApplicationType = ApplicationType.Client,
            };

            Console.WriteLine("Step 2 - Create a certificate store.");
            var certificateStore = new DirectoryStore(
                Path.Combine(Environment.GetFolderPath(Environment.SpecialFolder.LocalApplicationData), "Workstation.ConsoleApp", "pki"));

            Console.WriteLine("Step 3 - Create a session with your server.");
            var channel = new UaTcpSessionChannel(
                appDescription,
                certificateStore,
                ShowSignInDialog,
                discoveryUrl);
            try
            {
                await channel.OpenAsync();

                Console.WriteLine($"  Opened channel with endpoint '{channel.RemoteEndpoint.EndpointUrl}'.");
                Console.WriteLine($"  SecurityPolicyUri: '{channel.RemoteEndpoint.SecurityPolicyUri}'.");
                Console.WriteLine($"  SecurityMode: '{channel.RemoteEndpoint.SecurityMode}'.");
                Console.WriteLine($"  UserIdentity: '{channel.UserIdentity}'.");

                Console.WriteLine("Press any key to continue...");
                Console.ReadKey(true);

                // Prepare array of nodes to read.
                var registerNodesRequest = new RegisterNodesRequest
                {
                    NodesToRegister = new[]
                    {
                        NodeId.Parse("ns=3;s=\"TestData\".\"TestVector\".\"X\""),
                    }
                };

                var registerNodesResponse = await channel.RegisterNodesAsync(registerNodesRequest).ConfigureAwait(false);

                // Read the nodes.
                var readRequest = new ReadRequest
                {
                    NodesToRead = registerNodesResponse.RegisteredNodeIds.Select(
                        id => new ReadValueId { NodeId = id, AttributeId = AttributeIds.Value })
                    .ToArray()
                };

                Console.WriteLine("Press any key to end the program...");

                while (!Console.KeyAvailable)
                {
                    // Read the nodes.
                    var readResponse = await channel.ReadAsync(readRequest).ConfigureAwait(false);

                    // Write the results.
                    for (int i = 0; i < readRequest.NodesToRead.Length; i++)
                    {
                        Console.WriteLine($"{readRequest.NodesToRead[i].NodeId}; value: {readResponse.Results[i]}");
                    }

                    await Task.Delay(1000);
                }
                await channel.CloseAsync();
            }
            catch (ServiceResultException ex)
            {
                if ((uint)ex.HResult == StatusCodes.BadSecurityChecksFailed)
                {
                    Console.WriteLine("Error connecting to endpoint. Did the server reject our certificate?");
                }

                await channel.AbortAsync();
                throw;
            }
        }
```

LouisBaranger, 2021-12-20:
Thank you all,

I actually do a request (Set a bit at true) and waits for an answer (An other bit at true).
So the request answer bit is always true until I write false on the request but I get no notification of this change.

I will try to read with RegisterNodes later, I hope I can read all my 200 variables in less than 500ms.

closed 2023-03-16 by awcullen