convertersystems/opc-ua-client #218: Stack abstraction

quinmars, 2021-12-17:
In #111 we already discussed several ways to increase the testability of this library. I played with that idea in the last years, in particular, with the service providers approach. It took some time to find the right line to cut the code into pieces. But know I'm pretty confident that I found a good drawing line. So let's go into the details.

## Details
  
The stack profiles is a set of providers used by the channels to create the connection, the encryption and encoding classes. It will be passed to the constructor of the channel.
  
```csharp
    public class StackProfile
    {
        public IEncodingProvider EncodingProvider { get; }
        public ISecurityProtocolProvider SecureConversationProvider { get; }
        public ITransportConnectionProvider TransportConnectionProvider { get; }
        
        // constructor ...
    }
```

It's designed to be a constant. There will be only few instances for the different profiles. At the moment there is just one
instance for the BinaryUascTcp profile.

```csharp
    public static class StackProfiles
    {
        public static StackProfile BinaryUascTcp { get; }
            = new StackProfile(
                new BinaryEncodingProvider(),
                new UaSecureConversationProvider(),
                new UaTcpConnectionProvider()
            );
    }
```
More interestingly are the interfaces that are returned by the providers.

There is the `ITransportConnection` interface to establish a connection to a remote endpoint:
  
```csharp
    public interface ITransportConnection : IAsyncDisposable
    {
        Task<TransportChannelOptions> OpenAsync(uint protocolVersion, TransportChannelOptions localOptions, CancellationToken token);

        Task SendAsync(byte[] buffer, int offset, int count, CancellationToken token);
        Task<int> ReceiveAsync(byte[] buffer, int offset, int count, CancellationToken token);
    }
```

and the `ISecurityProtocol` interface to perform the message signing, encryption and decryption.
  
```csharp
    public interface ISecurityProtocol
    {
        uint ChannelId { get; set; }
        uint TokenId { get; set; }
        byte[]? RemoteNonce { get; set; }

        Task EncryptMessage(Stream bodyStream, uint messageType, uint requestHandle, Func<ArraySegment<byte>, Task> consume);
        Task<(uint messageType, uint requestHandle)> DecryptMessage(Stream bodyStream, Func<ArraySegment<byte>, Task<int>> recieve);
        byte[]? GetNextNonce();
        void UpdateKeys();
    }
```
  
## Things also changed

Since the channels are now no longer TCP specialize, I removed the leading `UaTcp`. I'm playing at the moment with a server
implementation, so I thought it'd be a good idea to add the word client to them, i.e., `ClientSessionChannel`
instead of `UaTcpClientSessionChannel`. If that ever turns out to be something
useful, we will not need to rename those channels again.

## Untouched

While it would be now possible to have an `ISecurityProtocol` implementation that does not employ bountycastle, I have not tried
to remove it from the `ICertificateStore` interface. I think that's a different goal that could be addressed at a later point. Same is
true for the SessionChannel.

## What next

You can find my current branch https://github.com/quinmars/opc-ua-client/tree/uastack. I do not know if it is really functional.
To be honest I haven't run the integration test. The aim was to find a good abstraction. If you like the approach, you could
create a new branch for it. I than would make smaller PRs, that are easier to review and of course I would take care that those
PRs will pass the integration tests.

awcullen, 2021-12-22:
I like the new interfaces and the greatly enhanced testability. I see that the profile can be specified in the constructor.
While there is only one profile atm, would you consider adding future profiles?  
Currently, when our client is opened, it connects to a server at an URL and calls GetEndpoints and the response tells us a TransportProfileUri  among other things.  

If endpoint starts opc.tcp://, then TransportProfileUri may be
http://opcfoundation.org/UA-Profile/Transport/uatcp-uasc-uabinary

if endpoint starts https://, then TransportProfileUri may be
http://opcfoundation.org/UA-Profile/Transport/https-uabinary
http://opcfoundation.org/UA-Profile/Transport/https-uasoapxml
http://opcfoundation.org/UA-Profile/Transport/https-uajson

if endpoint starts opc.wss://, then TransportProfileUri may be
http://opcfoundation.org/UA-Profile/Transport/wss-uasc-uabinary
http://opcfoundation.org/UA-Profile/Transport/wss-uajson

I would suggest we select a corresponding profile based the TransportProfileUri returned.  For now, we would support only uatcp-uasc-uabinary,  but in the future who knows

quinmars, 2021-12-22:
I'm very glad you like it. Indeed, the `StackProfile` class opens (intentionally) the door to implement several profiles. `wss-uasc-binary` should actually be some lines of code. At the moment the channels use the tcp stack as fallback if nothing or `null` is passed to the constructor. I also thought about guessing the stack profile on the basis of the url, the `TransportProfileUri` seems to be the better option. We probably should name the stack profiles to be close to the uri, i.e., instead of `StackProfiles.BinaryUascTcp` (as I did), `StackProfiles.UatcpUascUabinary` or `StackProfiles.TcpUascBinary`. Actually, I prefer the latter. Anyway, I think it should be possible to implement all known profiles with the presented interfaces. I haven't checked `https-uasoapxml` because it is deprecated.

quinmars, 2022-01-16:
@awcullen, do you know any freely accessible (OSS, shareware, trial version) sample server that supports profiles other than `opc.tcp`?

closed 2023-03-16 by awcullen