convertersystems/opc-ua-client #225: Private key of the certificate are not encrypted

FieryPhoenix100, 2022-01-17:
When creating certificates via DirectoryStore, it is not possible to encrypt the private key of the certificate. There is also no way to read the encrypted private key.

quinmars, 2022-01-18:
May I ask what you are trying to accomplish? The `DirectoryStore` can be used by the channel to retrieve the local certificate and the corresponding key parameters. Usually, you do not have to call the `DirectoryStore` methods yourself. There are cases were you might want to implement `IDirectoryStore`, for example when you want to add some user interface interactions. But even there I'd make it a wrapper around the existing `DirectoryStore`.

FieryPhoenix100, 2022-01-19:
I would like to be able to store private keys in encrypted form. I see that the keys are written to a file using PemWriter. PemWriter can encrypt keys. Unfortunately this feature is not available when using DirectoryStore. I think it's not safe to store private keys unencrypted.

awcullen, 2022-02-05:
I believe the suggestion is to add algorithm and password arguments (strings) to the DirectoryStore constructor (or have reasonable defaults).  Then use these parameters when reading and writing the private key files (pem encoded).

Anyone wish to contribute?