convertersystems/opc-ua-client #229: Is it possible to connect via https?

FieryPhoenix100, 2022-04-08:
I am trying to connect to the opc server using the opc.https protocol. When I try to do this, I get an error. Tell me is it possible to connect via the opc.https protocol?

awcullen, 2022-04-09:
We only support opc.tcp protocol at this time.  Sorry.

quinmars, 2022-04-25:
Recently, I changed the TCP/IP only code to potentially support other transport channels. Now, a websocket profile, for example, could be added with little effort. Unfortunately for `https` there would be more steps to go. The current implementation insists to establish a secure connection, where `https` relies on the connection encryption. Maybe we will add `https` support at some point, but at your position I would not wait for it.

closed 2022-05-06 by awcullen