convertersystems/opc-ua-client #238: Where is the v3.1.1 code?

lushess255, 2022-11-15:
I can't find the code used to build the v3.1.1 nuget package - there is no v3.1.1 tag.

For example, UaTcpSessionChannel is in the Workstation.UaClient nuget package but I couldn't find it any of the branches.

Where is the v3.1.1 code?  I'm sorry if there is simple, quick answer to this that I'm missing.

trivalik, 2022-12-16:
I guess one of these commits from 14. September 2021:

https://github.com/convertersystems/opc-ua-client/commit/f85050373b3753bcbb5889e97f4405c80a18da18
https://github.com/convertersystems/opc-ua-client/commit/49dba326237efcf5247a9efcd77854f5257290be
https://github.com/convertersystems/opc-ua-client/commit/8d4f8d5038cd09b7a254328bbd6ab812d80e84df

maybe the last one because it is flagged as verified. Detail you have to use a decompiler to check whether the change of commit is inside the dll from nuget package.

lushess255, 2022-12-20:
UaTcpSessionChannel is in the nuget package but I can't find its source on this site (including the commits you linked to).  Where is its source?

trivalik, 2022-12-21:
You can find it here https://github.com/convertersystems/opc-ua-client/blob/f85050373b3753bcbb5889e97f4405c80a18da18/UaClient/ServiceModel/Ua/Channels/UaTcpSessionChannel.cs (this is one commit from the 3 above, the hash in the link).
There was some changes in master. You cannot use github to search in old commits.