convertersystems/opc-ua-client #243: What is this mess?

dojo90, 2023-03-08:
There are now the `v3.1.0` and `v3.2.0` as a nuget package. Here in the repo there are no such tags. Nobody can look at the source code specifically for this tag. The last release is from November **2020** with tag `v3.0.0`.

In the last version (`v3.2.0`), the `UaTcpSessionChannel` class was removed from the project and no release notes were written about it. What is this shit?

ismdiego, 2023-03-08:
@dojo90, the UaTcpSessionChannel now it is called ClientTransportChannel. There have been a lot of changes during a refactor @trivalik did some months ago. Some readme files were updated, but obviously no release notes as you have just discovered.

I understand your frustration, specially if you are in a rush. Anyway, I recommend you to try to calm down a bit and politely ask for the changes and/or the Nuget v3.1.0 and v3.2.0 commits.

ismdiego, 2023-03-08:
Probable Nuget version vs repository status

v3.1.0: https://github.com/convertersystems/opc-ua-client/tree/59352073dd229e9e862c5a511173381c2d106b2d 
v3.1.1: https://github.com/convertersystems/opc-ua-client/tree/f85050373b3753bcbb5889e97f4405c80a18da18
v3.2.0: https://github.com/convertersystems/opc-ua-client/tree/11f54d92589610343a34cb2b4a38eca1736c870f

dojo90, 2023-03-08:
Thank you for your answers 👍

closed 2023-03-16 by awcullen