convertersystems/opc-ua-client #248: TranslateBrowsePathsToNodeIdsAsync now working

hudecekd, 2023-05-09:
Hello,

I am not able to make `TranslateBrowsePathsToNodeIdsAsync` to work. Is there something wrong with my code or is this bug in library?
I have tried to find some documentation for this method but without success.
I am trying to creaate nodeids for path from `root` which has `i=84` node id to its first child which is `Objects`.
I am getting status code `{0x80340000}` (BadNodeIdUnknown).

See code below. Could you advise? Thanks

```
var response = await channel.TranslateBrowsePathsToNodeIdsAsync(new TranslateBrowsePathsToNodeIdsRequest()
                { 
                    BrowsePaths =new BrowsePath[]
                    {
                        new BrowsePath()
                        {
                            StartingNode = new NodeId("i=84"),
                            RelativePath = new RelativePath()
                            {
                                Elements = new RelativePathElement[]
                                {
                                    new RelativePathElement()
                                    {
                                        TargetName = QualifiedName.Parse("Objects"),
                                        IncludeSubtypes = true,
                                        IsInverse = false,
                                        ReferenceTypeId = new NodeId(ReferenceTypeIds.HierarchicalReferences),
                                    }
                                }
                            }
                        }
                    },
                });
```

awcullen, 2023-05-17:
Hi.  In this library we use NodeID.Parse to get a NodeID from a string.  Used in two places below.

```
    new BrowsePath()
    {
        StartingNode = NodeId.Parse("i=84"),
        RelativePath = new RelativePath()
        {
            Elements = new RelativePathElement[]
            {
                new RelativePathElement()
                {
                    TargetName = QualifiedName.Parse("0:Objects"),
                    IncludeSubtypes = true,
                    IsInverse = false,
                    ReferenceTypeId = NodeId.Parse(ReferenceTypeIds.HierarchicalReferences),
                }
            }
        }
    }
```

closed 2023-05-17 by awcullen

hudecekd, 2023-05-17:
Now I see the difference when `new NodeId("i=84"),` or ` NodeId.Parse("i=84")` is used.
Thanks you very much for your help @awcullen