convertersystems/opc-ua-client #249: How to use X509Identity

fgasta, 2023-05-26:
Hi, 

i've triet to use the X509Identity, but I haven't found any documentation on how to load the certificate.

This is something that I've tried to come up with but of course it is not working:

`            var path = Path.Combine(Directory.GetCurrentDirectory(), "PKI", "own", "certs", "@1271480330916172243212159083043902389.der");
            var pathPrivate = Path.Combine(Directory.GetCurrentDirectory(), "PKI", "own", "private", "@1271480330916172243212159083043902389.key");

            var cert = new X509CertificateParser().ReadCertificate(File.ReadAllBytes(path));

            PemReader pr = new PemReader(new StringReader(pathPrivate));
            AsymmetricCipherKeyPair KeyPair = (AsymmetricCipherKeyPair)pr.ReadObject();
            RsaKeyParameters rsaParams = DotNetUtilities.ToRSAParameters((RsaPrivateCrtKeyParameters)KeyPair.Private);

            var key = new RsaKeyParameters(false, 1, 1);

            var channel = new ClientSessionChannel(
                clientDescription,
                certificateStore,
                new X509Identity(cert, rsaParams), 
                macchina.Indirizzo,
                SecurityPolicyUris.Basic256Sha256,
                loggerFactory
                );`

Any suggestion on the use of this type of identity?

Thank you!

awcullen, 2023-05-31:
I have used this code to make a X509Identity.  Hope this helps.

```
// read x509Identity
var userCert = default(X509Certificate);
var userKey = default(RsaKeyParameters);

var certParser = new X509CertificateParser();
var userCertInfo = new FileInfo(Path.Combine(pkiPath, "user", "certs", "ctt_usrT.der"));
if (userCertInfo.Exists)
{
    using (var crtStream = userCertInfo.OpenRead())
    {
        var c = certParser.ReadCertificate(crtStream);
        if (c != null)
        {
            userCert = c;
        }
    }
}
var userKeyInfo = new FileInfo(Path.Combine(pkiPath, "user", "private", "ctt_usrT.pem"));
if (userKeyInfo.Exists)
{
    using (var keyStream = new StreamReader(userKeyInfo.OpenRead()))
    {
        var keyReader = new PemReader(keyStream);
        if (keyReader.ReadObject() is AsymmetricCipherKeyPair keyPair)
        {
            userKey = keyPair.Private as RsaKeyParameters;
        }
    }
}
if (userCert != null && userKey != null)
{
    x509Identity = new X509Identity(userCert, userKey);
}

```

fgasta, 2023-07-10:
I've tried and your solution worked!
thank you!

closed 2023-07-10 by fgasta